Ultimaker/Cura #277: crash when adding a printer

greysAcademicCode, 2015-08-23:
Cura crashes when I add a printer. I've pulled and built everything fresh from github 10 min ago. Using the provided ubuntu build script:
```
Version: master
Platform: Linux-3.19.0-27-generic-x86_64-with-Ubuntu-15.04-vivid
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/UM/Qt/Bindings/AvailableMachinesModel.py", line 47, in createMachine
    app.setActiveMachine(app.getMachines()[index])
  File "/usr/local/lib/python3/dist-packages/UM/Application.py", line 261, in setActiveMachine
    self.activeMachineChanged.emit()
  File "/usr/local/lib/python3/dist-packages/UM/Signal.py", line 102, in emit
    func(dest, *args, **kargs)
  File "/home/grey/dev/Cura/cura/PrintInformation.py", line 188, in _onActiveMachineChanged
    self._updateTimeQualitySettings()
  File "/home/grey/dev/Cura/cura/PrintInformation.py", line 223, in _updateTimeQualitySettings
    setting_value = round(minimum_value + (maximum_value - minimum_value) * (self._time_quality_value / 100), options["precision"])
TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
```

nallath, 2015-08-24:
Master tends to be unstable, this is why we recommend users to only stick to the releases. I don't have this issue with the current version.

Why are you using master?

greysAcademicCode, 2015-08-24:
When I couldn't get stable to compile/fit together/work, I thought I'd try master.

greysAcademicCode, 2015-08-24:
On a slightly related note, the ubuntu build script provided in the stable branch of cura-build pulls from master for all the various Ultimaker repos, which seems kind of silly.

nallath, 2015-08-24:
It's sort of a catch all script that can do a ton of things. But if you can't get stable to work, log issues about that. I'd rather not spend time on getting master to work for you if you don't intend to do development work.

greysAcademicCode, 2015-08-24:
My intent here is to package it for Arch Linux.

nallath, 2015-08-24:
Then I suggest you try and build the 15.06 branch as that is our release branch. We will release 15.10 soon, which will also get it's own branch.

The issue you have seems to have something to do with the time slider, which requires certain settings to function, but it's unable to find those settings in the .json file that it uses. Could you check which one it's using?

ad1217, 2015-08-24:
Hello, I am also trying to build for Arch. 15.06 is not really an option, since there are some bugs with it and PyQT 5.5 (which seem to be fixed in master).
The error occurs seems to occur in all of the Ultimaker printers. I haven't tested any others.

greysAcademicCode, 2015-08-25:
Hi Adam. I've got it working. I maintain the `cura` package in the AUR which builds from the 15.06.03 source releases. Give it a shot and let me know if it's working for you, I've tested it with good success on a few different machines.

I see you maintain a few cura related packages in the AUR as well, If you'd like, you can add me (username greyltc in the AUR) as a co-maintainer and I can try to fix them up!

nallath, 2015-08-26:
The open beta will be released on 1 September (at which point we will create a 15.10 branch). If you guys want we could notify you if this happens so you can do the building for arch.

greysAcademicCode, 2015-08-27:
Thanks nallath, I'd appreciate a notification.

orbiterprof, 2015-08-27:
I am new to the forum but my cura 15.06.03 also crashes when I want to add a printer from the printer selection GUI.

Extremix, 2015-08-29:
Same as orbiterprof. 
in my windows 10 machine (my personnal computer) it works fine. But when I installed it in my professional computer **(windows 7) it crashes** when we select any machine except **Ultimaker Original**

Does it have to do with admin priveleges? Will you have a fix? Does old versions work fine ?

Ghostkeeper, 2015-08-29:
I haven't been able to reproduce this but machine settings are stored in %APPDATA% on Windows, so it shouldn't have anything to do with administrator privileges. Of course you can check that by running the program as administrator.

We are working to overhaul the settings system in Cura since that was the most pressing flaw of version 15.06, causing many crashes. Machine settings is a part of that. This work is going on in the [per_object_settings](https://github.com/Ultimaker/Cura/tree/per_object_settings) branch. Versions 15.04 and earlier work completely different and might not suffer that bug.

Extremix, 2015-08-29:
I think you are right, I can see a file created when I change settings for the **Ultimaker Original** so I dont think it is related to administrator privileges.

I will try to install version 15.04.2 for now (My professional computer, where I experience the bug, is not convinient for me for testing because I dont have the rights to install software by my own).

awhiemstra, 2015-08-31:
Crashing when selecting anything other than UMO is a known issue related to a combination of drivers + hardwre. See #96 for the details. As mentioned there, I implemented a workaround for it, starting with --disable-textures should fix the crashes, but means you will not get any texture on the build plate.

ad1217, 2015-08-31:
This crash is with the Ultimaker Original. All Ultimakers (and maybe others, I haven't tested) crash the same way. --disable-textures doesn't help. It also happens on both Intel and NVIDIA graphics cards. I don't think that this is the same bug.

awhiemstra, 2015-08-31:
What @Extremix mentioned is the same crash as in #96.

As for the other one, since you are on Arch, any chance you can provide a complete backtrace? I am also on Arch and everything works for me.

ad1217, 2015-08-31:
Not sure what you mean by "complete" backtrace, but here are my logs:
GUI:
```
Version: master
Platform: Linux-4.1.6-1-ARCH-x86_64-with-arch-Arch-Linux
Qt: 5.5.0
PyQt: 5.5

Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.4/site-packages/UM/Qt/Bindings/AvailableMachinesModel.py", line 47, in createMachine
    app.setActiveMachine(app.getMachines()[index])
  File "/usr/lib/python3.4/site-packages/UM/Application.py", line 261, in setActiveMachine
    self.activeMachineChanged.emit()
  File "/usr/lib/python3.4/site-packages/UM/Signal.py", line 102, in emit
    func(dest, *args, **kargs)
  File "/usr/lib/python3.4/site-packages/cura/PrintInformation.py", line 188, in _onActiveMachineChanged
    self._updateTimeQualitySettings()
  File "/usr/lib/python3.4/site-packages/cura/PrintInformation.py", line 223, in _updateTimeQualitySettings
    setting_value = round(minimum_value + (maximum_value - minimum_value) * (self._time_quality_value / 100), options["precision"])
TypeError: unsupported operand type(s) for -: 'NoneType' and 'NoneType'
```
Terminal:
```
2015-08-31 09:28:11,605 - INFO - Loaded plugin ConsoleLogger
2015-08-31 09:28:11,608 - INFO - Loaded plugin WireframeView
2015-08-31 09:28:11,612 - INFO - Loaded plugin MeshView
2015-08-31 09:28:11,618 - INFO - Loaded plugin LocalFileOutputDevice
2015-08-31 09:28:11,618 - WARNING - Plugin ConsoleLogger was already loaded
2015-08-31 09:28:11,622 - INFO - Loaded plugin STLWriter
2015-08-31 09:28:11,625 - INFO - Loaded plugin OBJWriter
2015-08-31 09:28:11,629 - INFO - Loaded plugin STLReader
2015-08-31 09:28:11,633 - INFO - Loaded plugin OBJReader
2015-08-31 09:28:11,666 - INFO - Checking for new version of cura
2015-08-31 09:28:11,668 - INFO - Loaded plugin UpdateChecker
2015-08-31 09:28:11,676 - INFO - Loaded plugin CameraTool
2015-08-31 09:28:11,695 - INFO - Loaded plugin RotateTool
2015-08-31 09:28:11,698 - INFO - Loaded plugin SelectionTool
2015-08-31 09:28:11,704 - INFO - Loaded plugin MirrorTool
2015-08-31 09:28:11,708 - INFO - Loaded plugin TranslateTool
2015-08-31 09:28:11,715 - INFO - Loaded plugin ScaleTool
2015-08-31 09:28:11,718 - INFO - Loaded plugin FileLogger
2015-08-31 09:28:11,723 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-08-31 09:28:11,728 - INFO - Loaded plugin ChangeLogPlugin
2015-08-31 09:28:11,760 - INFO - Loaded plugin CuraEngineBackend
2015-08-31 09:28:11,779 - INFO - Loaded plugin USBPrinting
2015-08-31 09:28:11,794 - INFO - Loaded plugin 3MFReader
2015-08-31 09:28:11,797 - INFO - Loaded plugin GCodeWriter
2015-08-31 09:28:11,802 - INFO - Loaded plugin SliceInfoPlugin
2015-08-31 09:28:11,808 - INFO - Loaded plugin LayerView
2015-08-31 09:28:11,821 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
2015-08-31 09:28:11,821 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
2015-08-31 09:28:11,821 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
2015-08-31 09:28:11,822 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
file:///usr/lib/qt/qml/QtQuick/Controls/ScrollView.qml:198: TypeError: Cannot assign to read-only property "progress"
2015-08-31 09:28:12,336 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
file:///usr/lib/qt/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/qt/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/qt/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/qt/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/share/cura/resources/qml/SidebarSimple.qml:128:22: Unable to assign [undefined] to bool
2015-08-31 09:28:12,455 - INFO - Started engine process: /usr/bin/CuraEngine
2015-08-31 09:28:12,681 - DEBUG - Backend connected on port 49674
2015-08-31 09:28:15,821 - DEBUG - Loaded a mesh with 10452 vertices
```
Also, how are you using it on Arch? Are you using a PKGBUILD, running it from the source, or something else?

Extremix, 2015-09-01:
Hi,

I just want to confirm that --disable-textures fixes the issue for me.

orbiterprof, 2015-09-01:
Hello,

How exactly did you do this?

My software programming skills end in MATLAB.

What steps did you take to resolve the crash. I will appreciate a detailed information.

Thanks.

Kelechi

------------------------------

Extremix, 2015-09-01:
one way to do it: Create a shortcut then modify its target properties

[image]

orbiterprof, 2015-09-02:
It worked!
Thanks a lot.
I am building the first 3D printing service company in West Africa. Think about  the African version of shapeways.
You may check us out. www.afera3d.com ( the site is still undergoing some rework). I will appreciate your input.
Thanks once again.
regards,
Kelechi
 --------------------------------------------------------------------------------- ...We can choose to either see the obstacles or the possibilities....kco

mikenew12, 2015-09-03:
I'm on OS X 10.10.5, and this issue is resolved for me as of 5e332c4cb2f02dba90d3ec26c7615b974a527dbc (not sure which commit in particular fixed it).

ad1217, 2015-09-03:
Also seems to work on Arch Linux now.

closed 2015-09-16 by nallath