Ultimaker/Cura #278: Combing

EstebanPacheco, 2015-08-23:
Hello I'be made an small test between Cura 15.04 and Cura 15.06.03, i don't know why the outter surface looks like this, i think in some way the combing function is not working:
First image>15.04
second image>15.03.06
[image]
[image]

thanks...

nallath, 2015-08-24:
@BagelOrb This looks like a cura-engine issue.

BagelOrb, 2015-08-24:
I cannot reproduce your exact problem without the STL and the settings.

However, I've just solved a couple of bugs concerning the outer walls. One bug resulted in warts and another in wavy skins. These bugs are now solved, so I expect your bug to be solved as well.

In the next release the whole combing mechanism has been overhauled. Computation is slower, but the results should be better.

closed 2015-08-24 by BagelOrb