Ultimaker/Cura #279: Retraction does not work when combing is enabled and gap is narrow

lawrencejohnston, 2015-08-23:
Version 15.06.03

When the gap between two parts is narrow (0.6mm in this case) retraction doesn't seem to happen when combing is on (though it works fine with it off). Retraction should occur because the two parts are not connected (so there's no way for combing to avoid the jump) and the gap is wider than the minimum retraction travel.

I've tried tweaking the minimum travel and minimal retraction, and even with both set to 0 this issue still occurs.

Retraction settings:
Enable: true
Speed: 40 mm/s
Distance: 4.5mm
Minimum travel: 0.5mm
Enable combing: true
Minimal Retraction: 0.02mm

[Scad, STL, and GCode (both with and without combing enabled)](https://gist.github.com/lawrencejohnston/a17b4ff78e18ee33cc86)

nallath, 2015-08-24:
@BagelOrb Looks like a Cura-engine issue.

BagelOrb, 2015-08-24:
Combing has been completely overhauled. It's very unlikely the same bug is still/again in there.

closed 2015-08-24 by BagelOrb