Ultimaker/Cura #283: Cura crashes on startup

TMHogenhout, 2015-08-24:
Application Version: Cura 15.06.03
 Platform: Windows 7
 Qt: << The version of Qt used >>
 PyQt: << The version of PyQt used >>
 Display Driver: Standard VGA graphics adapter

Steps to Reproduce:
Just start it 

Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 137, in _render
  File "c:\dev\uranium\UM\Qt\QtGL2Renderer.py", line 159, in beginRendering
  File "c:\dev\uranium\UM\Qt\QtGL2Renderer.py", line 326, in _initialize
AttributeError: 'NoneType' object has no attribute 'initializeOpenGLFunctions'
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 137, in _render
  File "c:\dev\uranium\UM\Qt\QtGL2Renderer.py", line 159, in beginRendering
  File "c:\dev\uranium\UM\Qt\QtGL2Renderer.py", line 326, in _initialize
AttributeError: 'NoneType' object has no attribute 'initializeOpenGLFunctions'

nallath, 2015-08-24:
This means the GL context could not be loaded. Does the machine have the right gl support?

awhiemstra, 2015-08-24:
"Display Driver: Standard VGA graphics adapter"

Sounds to me like it does not.

nallath, 2015-08-26:
Closing this as we simply require openGL. Nothing we can do about that.

closed 2015-08-26 by nallath