Ultimaker/Cura #291: Cura 15.06.03 not opening

Orphidia, 2015-08-25:
Cura does not seem to be opening. It has been working fine on the same computer until now. When trying to launch the program it comes up with the following error message:

Version: 15.06.03
Platform: Windows-8-6.2.9200
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 122, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1044, in _read
configparser.DuplicateSectionError: While reading from 'C:\\Users\\Orphidia/AppData/Local/cura\\cura.cfg' [line 18]: section 'backend' already exists

BagelOrb, 2015-08-25:
This is a known bug. Somehow a section in that file (C:\Users\Orphidia/AppData/Local/cura\cura.cfg) became duplicate. 
You can either delete that file or find the duplicate section and delete it.

Closing this, since its a duplicate..

closed 2015-08-25 by BagelOrb

Orphidia, 2015-08-25:
Thanks, for your help!!