Ultimaker/Cura #293: State of the master branch

fieldOfView, 2015-08-25:
I have managed to get the master branch running on Windows 7, but I am running into some strange behavior that I don't have in the official releases (15.06 branch). Before I put more time into investigating the peculiarities of my setup (I downloaded and fixed a binary version of protobuf instead of installing mingw and building it myself), are there known issues with master vs 15.06?

Some of the weirdness:
* In the print setup controls, floats are not properly rounded/displayed (eg: 0.10000000000000001 instead of 0.1)
* When I rotate an object, it ends up underneath the buildplate and doesn't reposition itself

awhiemstra, 2015-08-25:
"In the print setup controls, floats are not properly rounded/displayed (eg: 0.10000000000000001 instead of 0.1)"

This is a known issue with Qt 5.5. I do not know if this will be solved any time soon as I am not planning on switching Qt/PyQt versions just yet.

"When I rotate an object, it ends up underneath the buildplate and doesn't reposition itself"

This is a known bug due to the addition of cut off object bottom. I belive there is already a bug for it.

nallath, 2015-08-25:
The rotation of the object is a known issue; I recently implemented the cutoff distance again, which disables the colission of objects with the build plate (Which results in the model not being pushed above the build plate).

Also; on 1 september we have a feature freeze, at which point we will start dedicated bug squashing.

closed 2015-08-25 by nallath

fieldOfView, 2015-08-25:
Thanks, exactly what I needed to hear. My tinker-fu is still strong ;)