Ultimaker/Cura #294: Bug: Overlapping convex hulls

BagelOrb, 2015-08-26:
Expected: the shadows of objects signifying their convex hull cannot overlap due to the way collision detection is implemented.
Actual: overlap

[image]

nallath, 2015-08-26:
Seems to be caused by the bounding box intersection. The bounding box fits snugly over the object, whereas the convex hull is offset with a minkowsky hull (so it's sometimes larger).

BagelOrb, 2015-08-26:
You're begging the question.
What's the reason the convex hull is offsetted and the bounding box is not?!

nallath, 2015-08-26:
How is this begging the question?

BagelOrb, 2015-08-26:
My question is on the convex hulls overlapping; your answer simply restates that the convex hulls are overlapping and gives the cause. You're not giving a reason.

What's the reason the convex hull is offsetted and the bounding box is not?

nallath, 2015-08-26:
I am giving a reason. The reason is that the bounding boxes are not overlapping (which caused the intersection to be removed from the list to be checked, so it never intersected the hulls). I did not only re-state that they are overlapping. Even if I did, that is not begging the argument. 

To answer your second question; The convex hull states in what area other objects cannot be printed. The bounding box is there so we can see if the object can be printed on the build plate. I'm not 100% sure if that behaviour is correct, but that's how we use it.

BagelOrb, 2015-08-27:
I still don't totally understand **why**. Many people confuse reasons with causes - no worries.

As I understand it now the convex hulls are supposed to have an overlap because the convex hull ensures non-overlapping models by enforcing that a model doesn't overlap with the convex hull, not by enforcing that the convex hulls of the models are non-overlapping.

nallath, 2015-08-27:
No, convex hulls should never overlap. They had an overlap because the check never got to the point of checking the convex hulls (as it stopped at checking bounding box A with bounding box B).

BagelOrb, 2015-08-27:
So this is a bug.
You say convex hulls should never overlap, but here they do.

nallath, 2015-08-27:
Look at the commit and what i wrote. I explicitly stated that i already fixed this.

closed 2015-08-27 by nallath

BagelOrb, 2015-08-28:
Ohh ok. I'm sorry.