Ultimaker/Cura #296: Crash at startup

BagelOrb, 2015-08-26:
Just switched from branch 15.06 and killed the process last time

I keep experiencing this bug, so I cannot start Cura.

log
----
Version: master
Platform: Linux-3.19.0-26-generic-x86_64-with-Ubuntu-15.04-vivid
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 17, in <module>
    app.run()
  File "/home/tim/Development/Cura/cura/CuraApplication.py", line 169, in run
    self.setActiveMachine(machine)
  File "/home/tim/Development/Uranium/UM/Application.py", line 261, in setActiveMachine
    self.activeMachineChanged.emit()
  File "/home/tim/Development/Uranium/UM/Signal.py", line 102, in emit
    func(dest, *args, **kargs)
  File "/home/tim/Development/Uranium/UM/Scene/Platform.py", line 49, in _onActiveMachineChanged
    self.setMeshData(app.getMeshFileHandler().read(Resources.getPath(Resources.MeshesLocation, mesh), center = False).getMeshData())
AttributeError: 'NoneType' object has no attribute 'getMeshData'

awhiemstra, 2015-08-26:
Delete your config files in .local/share/cura/settings . The 15.06 files are incompatible with master and there is no cleanup yet.

closed 2015-08-26 by awhiemstra

BagelOrb, 2015-08-26:
Ah yes. However, without the config I encounter another bug which causes me to be unable to use Cura (https://github.com/Ultimaker/Cura/issues/288)