Ultimaker/Cura #299: Catch errors early

BagelOrb, 2015-08-26:
I'm getting the impression that any bug results in a total crash of the program.

Isn't there a way such that half of the exceptions are caught early without closing the whole application?

I'm imagining for example catching all visualization bugs in such a way that only the visualization system has to restart without closing the program.

I don't know much about the inner workings of the frontend, but I would like it if my workflow wouldn't be interrupted so much.

nallath, 2015-08-26:
If you want a stable release, stick with the release branches. The master is a nightly (and will be reasonably prone to crashes).

In most cases a visual bug should not lead to a crash (as these are caused by QML interpetation errors). 

At points where we expect certain exceptions to occur, we catch them, but there are just too many points of failure, which is why we implemented the catch all in the first place.

awhiemstra, 2015-08-26:
There is a behaviour change in 15.10 where any uncaught exception will trigger the crash handler and exit. With 15.06 it only triggers for exceptions during startup. This is intentional because otherwise the program can get into an undefined state where certain bits stop working. Additionally, making them very explicit means we get much more information about what went wrong instead of just "something isn't working quite right". This is similar behaviour to things like assertions in C++ that will error out - and often abort the program - when certain conditions occur.

As for the rest, the nature of an uncaught exception is that it is uncaught. This means it is an error somewhere else in the program, an error that we did not except and thus did not guard for. While it would be possible to brute-force catch all errors in certain places, the only thing this would do is hide the error and make it harder to debug. Additionally, suppose an error occurs while rendering a frame and we catch all errors while rendering a frame and simply drop the frame when an error occurs. There is no reason to assume the error condition will magically fix itself and not occur the second time we render a frame. So you end up never being able to render a frame and might as well just crash.

closed 2015-08-26 by awhiemstra

BagelOrb, 2015-08-27:
I think you misunderstood me.

I'm not proposing any such drastic change. I want exactly the current behaviour, including the error message, but without the whole program closing altogether.

If you principal arguments would still hold they would also hold for arguing that the whole computer should close down when an error occurs.

I'm just wondering whether it's possible to restart half of the application - the half where the error occurred - instead of closing the whole application. Is it possible to divide the progam in such a way?

Ghostkeeper, 2015-08-27:
At least the exceptions caused by plugins are caught, right?

nallath, 2015-08-27:
They should be caught. I've had situations where this failed though.

We could make a distinction between exceptions thrown by required plugins and non-required plugins. In the case of a non-required plugin a notification would suffice.

awhiemstra, 2015-08-27:
Exceptions from plugins thrown while loading the plugin are caught. Exceptions that happen later are not necessarily caught since they are interwoven into the rest of the system at that point.

BagelOrb, 2015-08-27:
Sounds like the plugins are too interwoven to catch their exceptions early and restart the plugin.
Oh the issue is already closed...