Ultimaker/Cura #300: Pass correct json file to backend

BagelOrb, 2015-08-26:
Steps:
- select UM2
- slice something
- open engine log

log
----
Calling engine with: ['/home/tim/Development/CuraEngine/build/CuraEngine', 'connect', '127.0.0.1:49675', '-j', '/home/tim/Development/Cura/cura/../resources/settings/fdmprinter.json', '-vv']
[...]

expected
----
.../ultimaker2.json

-------------------------------------------------

note: this is a change due to the fact that not all settings are in fdmprinter.json anymore.

nallath, 2015-08-26:
Would you only want the utlimaker2.json? This doesn't have all the settings in it (eg; the engine would be responsible for looking up the inherited files). If that's not a problem, this is an easy fix.

BagelOrb, 2015-08-26:
The engine now looks up the inherited files.

However, still not all settings are available. Notable the dual extrusion settings.

Please implement the easy fix :)

awhiemstra, 2015-08-26:
Except that it is not as simple as it seems. How does the engine look up the other files? Relative to the path of the intial JSON? Because I already have a situation where that fails. We now support loading machine defintiions from user directories, like ~/.local/share/cura/machines/ . This directory will not contain fdmprinter.json or similar "base" files. Instead, they are looked up through Uranium's resource system. So you end up with ~/.local/share/cura/machines/ultimaker2_modified.json as initial file and /usr/share/cura/machines/fdmprinter.json as base file. So unless you want to implement all of Uranium's resource lookup we will need to think of a proper way to handle cases like this.

BagelOrb, 2015-08-27:
I would say the best and easiest way is to have fdmprinter.json have ALL setting definitions.
The fact that the setting definitions are there should be independent from any handling of the settings.

As it is now, it is quite handy to have the inheritance implemented in the backend, even though it will only work for json files in the same dir; the backend uses the defaults of e.g. ultimaker2.json. As I see it, the backend usage doesn't really need to be very user friendly anyway ;)

Alternative: we could require that there is at least one json file in the "base" directory which - through inheritance - includes all settings. This one all-including json should then always be sent to the backend. Then dual extrusion settings will not be sent via the command socket, but the defaults will be taken from the all-including json file.

Thinking more on this I think this is a really good way to go. Agree?

PS:
tl;dr
always send dual_extrusion_settings.json and always have fdmprinter.json in the same dir.

BagelOrb, 2015-09-08:
"always send dual_extrusion_settings.json and always have fdmprinter.json in the same dir."
How is this going?
I think this should be fixed before the release.

This is a 5 min job, you guys!

awhiemstra, 2015-09-08:
As said previously, dual extrusion is not a feature for 15.10. So this will not be fixed before then.

BagelOrb, 2015-09-08:
This is not a feature for dual extrusion, the engine needs to know all settings. You can view this as the API of the engine. If you don't supply the full API unexpected behaviour might result. Dual extrusion settings might become enabled while printing with a single extrusion printer.

nallath, 2015-09-09:
That's bad behaviour from the engine. If no settings are provided the engine should default to sensible settings. It's a fair assumption that at this point most printers have a single extruder and the engine should therefore default to one

BagelOrb, 2015-09-09:
The defaults to which the engine defaults are described in the json files. You are describing exactly the reason why dual_extrusion_printer.json should always be passed to the engine.

CuraEngine officially requires that json file, so you should call CuraEngine with the right parameters. If you think this shouldn't be a requirement we can have a discussion, but you shouldn't abstain from conforming to the way in which CuraEngine should be called. I.e. implement it even though you think CuraEngine shouldn't work this way.

nallath, 2015-09-09:
If a setting is not found in a definition, you should also have behaviour. Turning dual extrusion on when no dual extrusion settings can be found is always wrong, regardless of what should be passed.

BagelOrb, 2015-09-09:
All conceptual discussion aside - I am saying that passing the incorrect json to CuraEngine currently results in undefined behaviour.

I advise against knowingly invoking undefined behaviour, even though it seems that the undefined behaviour coincides with normal behaviour for most of the possible runs of CuraEngine.

BagelOrb, 2015-09-30:
This problem is not solved.

nallath, 2015-09-30:
It's up to Arjen what to do with this, but the last statement about this was that the current behaviour is the correct one (hence this was closed, as that makes this issue a wont fix).

BagelOrb, 2015-10-21:
@awhiemstra We need to talk about this and decide on a solution before the release.

BagelOrb, 2016-04-26:
Fixed.

closed 2016-04-26 by BagelOrb