Ultimaker/Cura #304: surprise unwanted automatic scaling

red2black, 2015-08-27:
I've recently used Cura 15.06.02 and Ulitmaker 2 to make a part and once it was made I discovered it was undersize.
With a bit of investigating it looks like the part was automatically scaled to fit onto the build area when the STL file was imported into Cura.
- I would never want automatic scaling without asking for it by using a 'scale to fit' tool;
- There was no warning that the part had been scaled;
- When the scale tool is activated it says the scale is 100% in each direction, but this is 100% of the automatically scaled item, not of the original size. So I can't adjust the scale to correct the automatically introduced error.
Although the part did not fit on the build area when originally imported it would have been possible to rotate the item so that it could be printed.
I guess some might say this isn't a bug it's a feature, but if it's intentional then it's unwanted and unannounced behaviour.

nallath, 2015-08-27:
Duplicate of https://github.com/Ultimaker/Cura/issues/259 and https://github.com/Ultimaker/Cura/issues/142

closed 2015-08-27 by nallath

red2black, 2015-08-27:
Sorry, had searched for "scaling" and found nothing.

nallath, 2015-08-28:
You should also check for closed issues :) I found these by looking at all issues about scaling.

red2black, 2015-08-28:
Hmm. If I'd found a closed issue describing the same problem without it linking to an open issue then I would have had to open a new issue, since the problem does exist. In any case, thanks for your help and for pointing me in the right direction.