Ultimaker/Cura #305: Bug: text doesn't fit

BagelOrb, 2015-08-28:
For quite some settings, the setting name is abbreviated. I don't think this is desirable.
I see two possible solutions:
- adjust font size
- wrap around to new line

[image]

Note: also the Save gcode text doesn't fit.

BagelOrb, 2015-08-28:
BTW: why is there a small black line to the left of the save button?

nallath, 2015-08-28:
The new layout isn't fully implemented yet, so there are some issues with the layout.

BagelOrb, 2015-08-31:
So is the issue handled in the new layout?

nallath, 2015-09-01:
Yes

BagelOrb, 2015-09-01:
Just saw the new layout; it's still handled the same.
Tamara just told me she was going to address the issue by displaying the full setting name at the top of the mouse-over description box.

BagelOrb, 2015-09-03:
Why did you close this?
Was it by accident?

nallath, 2015-09-03:
No, but creating bugs of things that are still WIP is fairly useless. The current designs are on master for archive / backup reasons, not because they are seen as done.

BagelOrb, 2015-09-03:
I am just saying there is a bug in the current design. This is not useless; it provides you with the information that something needs to be done to resolve this. You can close it when it's done; you shouldn't close it because a lot is being done around the subject.

TMHogenhout, 2015-09-03:
This is fixed in the new release, but as Jaime says it's a WIP so it's not really usefull to keep issues around on WIP related issues 

https://github.com/Ultimaker/Cura/commit/02206c5540a2ab99fbe43a6a223b644a3228aad6

BagelOrb, 2015-09-03:
I close my issues when I know I've solved them, not when work on them is in progress. I think that is the correct way to do it.

Closing the issue means that it is not an issue (anymore). When you're working on an issue it's still an issue.

nallath, 2015-09-03:
No, closing means that there is no need to have this information pop in your face all the time (eg; We don't need to track this, as this never was a 'real' issue to begin with; it's a result of you not using a release version but a known unstable version)

BagelOrb, 2015-09-03:
When you see bugs in the unreleased code not as 'real' issues, then these issues will become real in the next release.

There is a need to have the information. It serves as a reminder that the issue needs to be solved.

BagelOrb, 2015-10-27:
This bug is still present in the master branch, into which the 15.10 branch was just merged.

Would this be the time to say 'I told you so'?

nallath, 2015-10-28:
I think the setting abbreviation is by design.

BagelOrb, 2015-10-28:
The setting abbreviation is indeed by design - that's why it's a UX bug.
It is bad user experience to not be able at all to see the full setting name.

Also it is very unhandy to help people with getting the settings right when they cannot see the full setting names. 

In person I mentioned a different solution: add the setting title in the yellow pop-up box.
I think I remember we then agreed upon that solution, but I'm not sure.

TMHogenhout, 2015-10-28:
1. it's fixed (to long ago to remember how and where)
2. this is going to change again entirely in the new design 
[image]

nallath, 2015-10-28:
I think bagelorb meant the setting names (this is why you should make two issues if there are two problems)

BagelOrb, 2015-10-29:
It clearly says so in the first sentence of the issue, but I understand how the save button in the picture grabs the attention.

thopiekar, 2016-09-17:
Doesn't exist anymore..

closed 2016-09-17 by thopiekar