Ultimaker/Cura #306: Moving parts problem and not able to multiple objects..

Moriteourious, 2015-08-30:
Video will tell more the a words so let me start with those:

https://www.dropbox.com/s/8t0sa8cm16cxdhm/2015-08-30%2021.05.40.mp4?dl=0
https://www.dropbox.com/s/72qphgysy0366xr/2015-08-30%2021.07.28.mp4?dl=0

..so everytime I try to move an object it, by defaults, go somewhere far beyond working area.. so I need to move A LOT to move it back to a place.. and then some more to have it where I need it to be. Sometimes it goes so far that my screen wont allow me to make it back to a plate even...

Also I have some problem with multiple items - you can also see at the movies that those are too much and Cura will not calculate time and material needed to print etc. Also I can not save it to file...
And I dont belive that those are THAT much to do at once..?

Ghostkeeper, 2015-08-31:
This bug is a duplicate of #100 so I'm going to change the topic and re-use this issue for the second bug you mentioned. I'm not sure on the status of issue #100 (the colleagues that are responsible aren't at work yet).

On the problem with multiple items, I know that @awhiemstra has been working on per-object settings and this might change some of that stuff around. He'll make sure it works proper before merging that stuff to master.

The rotating is a valid concern. Duplicates should also duplicate the transformation quaternions (including scale and rotation), but then find a new position on the build plate.

Ghostkeeper, 2015-08-31:
Issue #100 is not fixed yet, but the cause is known and it is intended to be fixed for version 15.10.

closed 2015-09-02 by nallath