Ultimaker/Cura #308: Cura Unable to Startup

thisismyacct, 2015-08-31:
Trying to launch the latest version of Cura (15.06.03) on Windows 7 Pro 64 on an Intel Celeron N2930 with Intel HD graphics.  The log has:

Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 137, in _render
  File "c:\dev\uranium\UM\Qt\QtGL2Renderer.py", line 159, in beginRendering
  File "c:\dev\uranium\UM\Qt\QtGL2Renderer.py", line 326, in _initialize
AttributeError: 'NoneType' object has no attribute 'initializeOpenGLFunctions'

I downloaded Cura_15.04.2 and installed it and it runs with no problem.

Ghostkeeper, 2015-08-31:
This is a duplicate of issue #283. Your video driver doesn't support OpenGL. OpenGL is absolutely required for Cura and it's impossible for us to deliver such driver updates along with our installer or anything.

You can try to update your video drivers. I'll close this for now since it is probably a problem on your end that we can't fix. If you've tried updating your video drivers and the problem persists, maybe it can be reopened.

closed 2015-08-31 by Ghostkeeper

awhiemstra, 2015-08-31:
Specifically, it is lacking OpenGL 2 support which is why 15.04 works but 15.06 does not. For 15.10 we will investigate a fallback for machines that lack OpenGL 2 support.