Ultimaker/Cura #309: Wire feed problem Prusa i3 with cura

11argosy, 2015-09-01:
everything loads up OK, SD card inserted print file chosen but as soon as the print head gets to the bed to start printing the printer unloads the wire/filament and will not reload, I assume it is a software problem as it was ok until Friday.

nallath, 2015-09-01:
Well, Cura hasn't updated since Friday if you're using one of the official releases.

11argosy, 2015-09-01:
I am using an official version and I have uninstalled this and re-installed but to no avail, my friend thinks its a print profile issue but I don't know how to check this.  John

11argosy, 2015-09-01:
By the way, thanks Nallath

nallath, 2015-09-01:
If it worked friday with Cura and now it doesn't, there is little I can do to help. Cura didn't change, so it's probably the machine.

closed 2015-09-01 by nallath

11argosy, 2015-09-01:
Thanks Jaime I May have the problem solved but will know more tomorrow,thanks for your help, regards John

BagelOrb, 2015-09-02:
This sounds like a firmware issue.