Ultimaker/Cura #313: BUG: No configuration wizard launch for Ultimaker Original +

jambalyle, 2015-09-02:
built a new Original +, but the configuration wizard is launching when I open up Cura. I'm new to the Ultimaker side of things, but from what I read it's supposed to walk me through calibrating the machine. or am I missing something?

nallath, 2015-09-02:
The old cura (<15.04) had a wizard. The version that will go into beta this week should have the wizard again. 15.06 and upwards are a complete rework of Cura, so not all features are there yet.

closed 2015-09-02 by nallath

Dim3nsioneer, 2018-07-02:
@Ghostkeeper : That wizard which was reintroduced disappeared from 3.2 to 3.3. Was that on purpose?

fieldOfView, 2018-07-02:
It is still there for me on 3.4

Dim3nsioneer, 2018-07-02:
with the checks? Not only the bed leveling...

Ghostkeeper, 2018-08-13:
Yeah it was sort of on purpose. We "moved responsibility to the user" to check your endstops and such. In other words, we removed it because it was too little value for too much maintenance.

Bed levelling should still be in there.

Dim3nsioneer, 2018-08-13:
In this case would you (or someone else at Ultimaker) mind to modify the ressource page accordingly (aka finishing the change)? It still says "When Cura asks you to check to press the left X endstop..."
https://ultimaker.com/en/resources/183-configuration
Thx.