Ultimaker/Cura #314: Bed is a bit warped

a4jp-com, 2015-09-02:
Can you add a setting to check more bed points if the bed is slightly warped? The wizard only checks 3 points now. Even an advanced setting to do this would be nice.

nallath, 2015-09-02:
The bed point checking is just for leveling, not for checking if it's warped. 

Did you have this problem with an Ultimaker? We could look into making an plugin that does this, but it probably wont have priority.

a4jp-com, 2015-09-02:
Yes. 

A plugin like that would be fantastic. Everything else is great with your program I just need to fix this problem.

nallath, 2015-09-02:
How did you get a bed to warp with an Ultimaker? Thinking a bit more on this; even if we have the warp of the platform, we don't have any software in place that could compensate for this. It's a much harder problem than just finding what the warp is.

a4jp-com, 2015-09-02:
I'm not sure, the glass is just a tiny bit higher near the center and lower near the edges for some reason. Any plugin or idea would be great.

nallath, 2015-09-02:
This would take a lot of time to implement. You're better off getting a proper (flat) bed.

closed 2015-09-02 by nallath