Ultimaker/Cura #316: UI: Per-object settings button out of place

fieldOfView, 2015-09-03:
The newly added per-object settings tool button is a bit out of place in the current toolbar. Mirror, scale and rotate each manipulate the geometry of the model. Per-object settings does not. Putting a little bit of extra space between the geometry manipulation tools and other tools would make sense.

A more logical order would be:
Scale
Rotate
Mirror
--
Per-object settings

BagelOrb, 2015-09-04:
Very valid point, though I am still in doubt about your solution.

nallath, 2015-09-04:
Not quite sure if we have an easy solution for this. The tool panel is auto generated based on the loaded plugins. This would require extra features (eg; the ability to define a type of a tool, by which they will be sorted)

fieldOfView, 2015-09-04:
I see now that the order of the tools is alphabetical. Not ideal, but not the most critical of issues right now. Defer to next version, unless one of us comes up with a brilliant, architecturally clean solution?

nallath, 2015-09-04:
Any solution i can come up with now requires new features, which means it's going to be next release.

tzminversion, 2015-09-13:
Does this enhancement mean you can change the print order of the items on the bed?

Ghostkeeper, 2015-09-13:
No, its intention was just to change the positions of the buttons in the user interface, not to affect the print job in any way.

fieldOfView, 2016-11-09:
Issue can be closed

Ghostkeeper, 2016-11-10:
I disagree. This has not been fixed yet.

BagelOrb, 2016-11-10:
This is coincidentally the exact same thing I said during the CCB yesterday. Opening per-object settings shouldn't deactivate the move tool etc.

BagelOrb, 2016-11-10:
I also had a totally different idea though:
- right click an object to say you want to edit per-object settings
- then it appears as another item in the print core setting tabs
- select the tab and all settings are editable.
- more link icons show up, because some settings are per-extruder, so changing the per-object setting should also change the extruder setting (or just make those settings greyed out)

I think this approach makes the per-object settings work flow a whole lot less awkward.

What do you think, Oh great Aldo?

nickthetait, 2016-11-10:
I agree that this would be an improved work flow. Another improvement it could allow is highlighting (in a different color?) those settings which have been overridden from the global profile.

fieldOfView, 2016-11-10:
I still think *this* issue can be closed; to quote GhostKeeper: "its intention was just to change the positions of the buttons in the user interface".  We probably need a new issue to discuss a rework of the whole tools system, including Per Model Settings.

The current Per Model Setting tool is a bit awkward. It could be interesting to explore switching the sidebar between settings for all models vs settings for the selected models. I did mull about that a while back, but never got the time to think much about it. The sidebar background could turn a shade when an object with Per Model Settings is selected.

The transformation tools could do with being unified; there's no good reason for Translate/Scale/Rotate being modal. Many other applications have unified/combined widgets that work just as well.

BagelOrb, 2016-11-11:
Fair enough, but I agree with @Ghostkeeper that this sissue is not implemented.
There is no visual separator between the per-object settings and the other tools.

BagelOrb, 2017-07-21:
I still think the per-object settings menu location needs some more love.

ianpaschal, 2018-02-27:
This hasn't been touched in half-a-year+, is pretty minor, and isn't "broken" so I'm closing it.

closed 2018-02-27 by ianpaschal