Ultimaker/Cura #317: Cura 15.06.03 Does not Slice, Can not save

lilmac024, 2015-09-03:
The model will not slice and as a result I can't save it to my SD card, it won't calculate any time, and won't show any layers in the layer view.

[image]

nallath, 2015-09-04:
Could you post your log file?

lilmac024, 2015-09-04:
The engine log appears totally empty.  

[image]

lilmac024, 2015-09-04:
I have uninstalled and reinstalled 2 times as well.

lilmac024, 2015-09-04:
I think I may have found the issue, it appears that I'm actually missing the CuraEngine.exe.  Will install that and see if it's resolved.

[image]

lilmac024, 2015-09-04:
problem resolved...my antivirus was deleting the CuraEngine.exe when it was turned on.  If I turn off the antivirus and install Cura it works perfectly.

closed 2015-09-04 by lilmac024

nallath, 2015-09-04:
Yeah, we've heard about this before. Most anti-viruses don't like local sockets (which we use to communicate). Thanks for posting the solution.