Ultimaker/Cura #319: 15.09.80: Bug report

CCS86, 2015-09-04:
Hi guys,

Just tested out the new version and am excited to see the direction Cura is moving. I really like the new features and layout. Hopefully I can help squash some bugs. I wasn't sure whether each of these should be posted as a separate issue, or if the whole picture was better.

After initially slicing a model quickly, switching to layer view revealed only shell paths, no infill generated. I had enabled the infill line distance setting, and set it to 0 for dense infill. I tried changing that to 2mm, and no new layers were generated. Even moving the part showed layers in the old position. I saved to a gcode file, then exited Cura. It crashed on exit, and displayed this message:

Version: 15.09.80
Platform: Windows-8-6.2.9200
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 150, in hideEvent
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 182, in windowClosed
  File "c:\dev\uranium\UM\Preferences.py", line 81, in writeToFile
  File "c:\dev\python34\lib\configparser.py", line 1208, in __setitem__
  File "c:\dev\python34\lib\configparser.py", line 1167, in set
  File "c:\dev\python34\lib\configparser.py", line 872, in set
  File "c:\dev\python34\lib\configparser.py", line 382, in before_set
ValueError: invalid interpolation syntax in "b'\\x00\\x00\\x00\\xbe\\x00\\x00\\x00\\x04\\xff\\xff\\xff\\xff\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x000file:///C:/Program%20Files%20(x86)/Cura_15.09.80\\xff\\xff\\xff\\xff\\x00\\x00\\x00\\x01'" at position 113

I reopened, and found that while it saved both machines that I created, and a "modified" high quality profile, the settings which I had enabled were not there. I tried to delete one of the machines, and after answering the confirmation to delete, nothing happens. Both remain.

I tried to load the gcode file, and this is where it gets weird. In Cura's load dialog box, after putting file type to any, there is my file: a.gcode, with a reasonable size. If I browse to that directory in Windows, the file is not there! If I right click the file from Cura's open dialog, and open it. notepad says it doesn't exist!

[image]

[image]

I am running Windows 10, i7-4810mq, 32 GB RAM.

CCS86, 2015-09-04:
Interesting. I reloaded the model and just selected the default high quality. It sliced and created the sparse infill. But, everywhere that needed 100% (top and bottom) was empty. If I set infill density to 100%, no infill at all.

[image]

CCS86, 2015-09-04:
Log data:

2015-09-03 19:46:08,800 - INFO - Loaded plugin FileLogger
2015-09-03 19:46:08,815 - INFO - Loaded plugin GCodeWriter
2015-09-03 19:46:08,834 - INFO - Loaded plugin LayerView
2015-09-03 19:46:08,854 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-03 19:46:08,874 - INFO - Loaded plugin PostProcessingPlugin
2015-09-03 19:46:08,904 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-03 19:46:08,950 - INFO - Loaded plugin SliceInfoPlugin
2015-09-03 19:46:08,973 - INFO - Loaded plugin CameraTool
2015-09-03 19:46:09,004 - INFO - Loaded plugin MirrorTool
2015-09-03 19:46:09,022 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-03 19:46:09,069 - INFO - Loaded plugin RotateTool
2015-09-03 19:46:09,101 - INFO - Loaded plugin ScaleTool
2015-09-03 19:46:09,116 - INFO - Loaded plugin SelectionTool
2015-09-03 19:46:09,163 - INFO - Loaded plugin TranslateTool
2015-09-03 19:46:09,194 - INFO - Checking for new version of cura
2015-09-03 19:46:09,194 - INFO - Loaded plugin UpdateChecker
2015-09-03 19:46:09,253 - INFO - Loaded plugin USBPrinting
2015-09-03 19:46:09,296 - INFO - Loaded plugin MeshView
2015-09-03 19:46:09,330 - INFO - Loaded plugin WireframeView
2015-09-03 19:46:09,881 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 19:46:09,901 - WARNING - Tried to set value of non-user setting layer_height
2015-09-03 19:46:09,901 - WARNING - Tried to set value of non-user setting layer_height
2015-09-03 19:46:10,687 - DEBUG - Loaded theme C:\Program Files (x86)\Cura_15.09.80\resources\themes\cura
2015-09-03 19:46:24,832 - DEBUG - Backend connected on port 49674
2015-09-03 19:47:53,930 - DEBUG - Loaded a mesh with 421560 vertices
2015-09-03 19:47:54,680 - ERROR - Job <cura.ConvexHullJob.ConvexHullJob object at 0x13C21B30> caused the exception:'NoneType' object has no attribute 'getSettingValue'
2015-09-03 19:47:54,681 - ERROR -   File "c:\dev\python34\lib\threading.py", line 889, in _bootstrap
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\uranium\UM\JobQueue.py", line 122, in run

2015-09-03 19:50:21,141 - DEBUG - Loaded a mesh with 10452 vertices
2015-09-03 20:00:35,613 - DEBUG - Killing engine process
2015-09-03 20:00:35,615 - INFO - Backend crashed or closed. Restarting...
2015-09-03 20:00:35,923 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 20:00:35,970 - DEBUG - Backend connected on port 49674
2015-09-03 20:00:42,116 - WARNING - Profile has error values. Aborting slicing
2015-09-03 20:00:44,489 - DEBUG - Killing engine process
2015-09-03 20:00:44,491 - INFO - Backend crashed or closed. Restarting...
2015-09-03 20:00:44,795 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 20:00:44,827 - DEBUG - Backend connected on port 49674
2015-09-03 20:01:20,161 - DEBUG - Killing engine process
2015-09-03 20:01:20,163 - INFO - Backend crashed or closed. Restarting...
2015-09-03 20:01:20,474 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 20:01:20,515 - DEBUG - Backend connected on port 49674
2015-09-03 20:01:21,920 - DEBUG - Killing engine process
2015-09-03 20:01:21,922 - INFO - Backend crashed or closed. Restarting...
2015-09-03 20:01:22,234 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 20:01:22,293 - DEBUG - Backend connected on port 49674
2015-09-03 20:01:36,274 - DEBUG - Killing engine process
2015-09-03 20:01:36,277 - INFO - Backend crashed or closed. Restarting...
2015-09-03 20:01:36,576 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 20:01:36,618 - DEBUG - Backend connected on port 49674
2015-09-03 20:02:12,503 - DEBUG - Killing engine process
2015-09-03 20:02:12,505 - INFO - Backend crashed or closed. Restarting...
2015-09-03 20:02:12,803 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 20:02:12,841 - DEBUG - Backend connected on port 49674
2015-09-03 20:02:30,999 - DEBUG - Killing engine process
2015-09-03 20:02:31,001 - INFO - Backend crashed or closed. Restarting...
2015-09-03 20:02:31,303 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 20:02:31,344 - DEBUG - Backend connected on port 49674
2015-09-03 20:03:12,860 - WARNING - Profile has error values. Aborting slicing
2015-09-03 20:03:35,501 - WARNING - Profile has error values. Aborting slicing
2015-09-03 20:03:38,224 - WARNING - Profile has error values. Aborting slicing
2015-09-03 20:03:42,690 - WARNING - Profile has error values. Aborting slicing
2015-09-03 20:05:13,802 - WARNING - Profile has error values. Aborting slicing
2015-09-03 20:05:19,199 - WARNING - Profile has error values. Aborting slicing
2015-09-03 20:06:12,570 - WARNING - Profile has error values. Aborting slicing
2015-09-03 20:06:36,455 - WARNING - Profile has error values. Aborting slicing
[... 134 lines cut ...]
2015-09-03 20:42:24,690 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.09.80\CuraEngine.exe
2015-09-03 20:42:24,829 - WARNING - Tried to set value of non-user setting retraction_hop
2015-09-03 20:42:24,830 - WARNING - Tried to set value of non-user setting adhesion_type
2015-09-03 20:42:24,830 - WARNING - Tried to set value of non-user setting infill_sparse_density
[... cut ...]
2015-09-03 20:42:24,831 - WARNING - Tried to set value of non-user setting speed_wall
2015-09-03 20:42:24,831 - WARNING - Tried to set value of non-user setting layer_height
2015-09-03 20:42:24,832 - WARNING - Tried to set value of non-user setting layer_height
2015-09-03 20:42:24,834 - DEBUG - Backend connected on port 49674
2015-09-03 20:42:25,439 - DEBUG - Loaded theme C:\Program Files (x86)\Cura_15.09.80\resources\themes\cura
2015-09-03 20:42:51,452 - WARNING - No active machine found when trying to save setting visibility

BagelOrb, 2015-09-04:
I don't know about the ghost gcode file, but maybe @Ghostkeeper can help ;)

About the ghost infill:
The top/bottom skin is not vizualized properly. It's showing the surrounding area as polygon instead of showing the polygon filled. This is also related to the 100% infill issue you mention; when you select 100% infill, the infill is replaced by skin, so that it doesn't subdivide the wholly filled areas into top/bottom skin and inner infill.

Ghostkeeper, 2015-09-04:
Seems to me that the ghost gcode file is due to Windows' windows not updating frequently enough. And python has done some weird things when closing while it is writing files, too.

There aren't any files in the cage where I keep my ghosts.

BagelOrb, 2015-09-07:
The not-slicing issue is an issue which is already reported in other github issues.

I'm viewing the ghost file as a fluke, so I'm closing this.

closed 2015-09-07 by BagelOrb