Ultimaker/Cura #320: SIGSEGV at closing

Ghostkeeper, 2015-09-04:
I just received a message that a segfault occurred, after I closed Cura. I'll post the details here.

Version: 15.09.80
Platform: Ubuntu 15.04
Qt: 5.4.1
PyQt: 5.4.1

Here is a logfile of everything I did with the new version:

    2015-09-04 09:19:03,206 - INFO - Loaded plugin FileLogger
    2015-09-04 09:19:03,218 - INFO - Loaded plugin OBJWriter
    2015-09-04 09:19:03,230 - INFO - Loaded plugin OBJReader
    2015-09-04 09:19:03,240 - INFO - Loaded plugin STLWriter
    2015-09-04 09:19:03,249 - INFO - Loaded plugin STLReader
    2015-09-04 09:19:03,249 - WARNING - Plugin ConsoleLogger was already loaded
    2015-09-04 09:19:03,344 - INFO - Loaded plugin CuraEngineBackend
    2015-09-04 09:19:03,349 - INFO - Loaded plugin SliceInfoPlugin
    2015-09-04 09:19:03,359 - INFO - Loaded plugin GCodeWriter
    2015-09-04 09:19:03,364 - INFO - Loaded plugin ChangeLogPlugin
    2015-09-04 09:19:03,371 - INFO - Loaded plugin RemovableDriveOutputDevice
    2015-09-04 09:19:03,423 - INFO - Loaded plugin USBPrinting
    2015-09-04 09:19:03,432 - INFO - Loaded plugin 3MFReader
    2015-09-04 09:19:03,446 - INFO - Loaded plugin LayerView
    2015-09-04 09:19:03,448 - DEBUG - Attempting to connect to /dev/ttyUSB0
    2015-09-04 09:19:03,477 - WARNING - Tried to set value of non-user setting layer_height
    2015-09-04 09:19:03,478 - WARNING - Tried to set value of non-user setting layer_height
    2015-09-04 09:19:03,512 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
    2015-09-04 09:19:03,512 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
    2015-09-04 09:19:03,513 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
    2015-09-04 09:19:03,513 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
    2015-09-04 09:19:04,133 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
    2015-09-04 09:19:04,294 - INFO - Started engine process: /usr/bin/CuraEngine
    2015-09-04 09:19:04,304 - DEBUG - Backend connected on port 49674
    2015-09-04 09:19:05,132 - INFO - Could not establish connection on /dev/ttyUSB0: 'Timeout'. Device is not arduino based.
    2015-09-04 09:19:57,692 - ERROR - Baud rate detection for /dev/ttyUSB0 failed
    2015-09-04 09:19:57,692 - WARNING - Printer connection state was not changed
    2015-09-04 09:19:57,692 - WARNING - Printer connection state was not changed
    2015-09-04 09:21:04,807 - WARNING - Unable to read file 
    2015-09-04 09:21:21,141 - DEBUG - Loaded a mesh with 88152 vertices
    2015-09-04 09:21:21,534 - ERROR - Job <cura.ConvexHullJob.ConvexHullJob object at 0x7f8e3838b978> caused the exception:'NoneType' object has no attribute 'getSettingValue'
    2015-09-04 09:21:21,540 - ERROR -   File "/usr/lib/python3.4/threading.py", line 888, in _bootstrap
        self._bootstrap_inner()
      File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
        self.run()
      File "/usr/lib/python3/dist-packages/UM/JobQueue.py", line 122, in run
        Logger.log("e", "".join(traceback.format_stack()))

    2015-09-04 09:42:28,919 - INFO - Loaded plugin FileLogger
    2015-09-04 09:42:28,922 - INFO - Loaded plugin OBJWriter
    2015-09-04 09:42:28,925 - INFO - Loaded plugin OBJReader
    2015-09-04 09:42:28,929 - INFO - Loaded plugin STLWriter
    2015-09-04 09:42:28,932 - INFO - Loaded plugin STLReader
    2015-09-04 09:42:28,933 - WARNING - Plugin ConsoleLogger was already loaded
    2015-09-04 09:42:28,992 - INFO - Loaded plugin CuraEngineBackend
    2015-09-04 09:42:28,996 - INFO - Loaded plugin SliceInfoPlugin
    2015-09-04 09:42:29,000 - INFO - Loaded plugin GCodeWriter
    2015-09-04 09:42:29,004 - INFO - Loaded plugin ChangeLogPlugin
    2015-09-04 09:42:29,011 - INFO - Loaded plugin RemovableDriveOutputDevice
    2015-09-04 09:42:29,025 - INFO - Loaded plugin USBPrinting
    2015-09-04 09:42:29,033 - INFO - Loaded plugin 3MFReader
    2015-09-04 09:42:29,039 - INFO - Loaded plugin LayerView
    2015-09-04 09:42:29,041 - DEBUG - Attempting to connect to /dev/ttyUSB0
    2015-09-04 09:42:29,141 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
    2015-09-04 09:42:29,141 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
    2015-09-04 09:42:29,142 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
    2015-09-04 09:42:29,142 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
    2015-09-04 09:42:29,261 - INFO - Started engine process: /usr/bin/CuraEngine
    2015-09-04 09:42:29,669 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
    2015-09-04 09:42:29,815 - DEBUG - Backend connected on port 49674
    2015-09-04 09:42:30,669 - INFO - Could not establish connection on /dev/ttyUSB0: 'Timeout'. Device is not arduino based.
[... 76 lines cut ...]
    Stack memory exhausted (SP below stack segment)
SegvReason: executing unknown VMA
Signal: 11
SourcePackage: cura
Stacktrace:
    #0 0x000007f514437d099 in ?? ()
    No symbol table info available.
    #1 0x00007f5141921f10 in ?? ()
    No symbol table info available.
    #2 0x0000000003174e70 in ?? ()
    No symbol table info available.
    #3 0x00007f5141921f50 in ?? ()
    No symbol table info available.
    #4 0x00000000031751d0 in ?? ()
    No symbol table info available.
    #5 0x00007fffefeac42f in ?? ()
    No symbol table info available.
    #6 0x00007f5151123b0e in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
    No symbol table info available.
    Backtrace stopped: previous frame inner to this frame (corrupt stack?)
Tags: third-party-packages vivid
Uname: Linux 3.19.0-26-generic x86_64
UnreportableReason: This is not an official Ubuntu package. Please remove any third party package and try again.
UpgradeStatus: No upgrade log present (probably fresh install)
UserGroups: adm cdrom dialout dip lpadmin plugdev sambashare sudo

Ghostkeeper, 2015-09-04:
I think libQt5Core is at fault.

nallath, 2015-09-04:
Yeah, seems that way. I've fixed the error caused by the convex hull job though.

nallath, 2015-09-15:
Upstream issue, nothing we can do about it.

closed 2015-09-15 by nallath