Ultimaker/Cura #324: Bug after saving gcode

BagelOrb, 2015-09-04:
branch: 15.10

steps
----
- Load model
- Let it slice
- at this point dragging triggers a reslice
- save gcode via File>Save All>Save to File
- see error log below
- get message that it has been saved

result
----
Dragging doesn't trigger reslice

log
----
2015-09-04 10:46:36,714 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
2015-09-04 10:46:36,714 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
2015-09-04 10:46:36,714 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
2015-09-04 10:46:36,714 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
2015-09-04 10:46:37,774 - DEBUG - Loaded theme /home/tim/Development/Cura/cura/../resources/themes/cura
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
2015-09-04 10:46:38,346 - INFO - Started engine process: /home/tim/Development/CuraEngine/build/CuraEngine
2015-09-04 10:46:38,351 - DEBUG - Backend connected on port 49675
2015-09-04 10:46:47,235 - DEBUG - Loaded a mesh with 312234 vertices
2015-09-04 10:48:25,717 - ERROR - [Errno 4] Interrupted system call
2015-09-04 10:48:41,550 - DEBUG - Writing to Local File /home/tim/Documents/3D_models/gcode/rabbit_x2_normal.gcode in text mode
tim@um-mba-173:~/Development/Cura$ PYTHONPATH=../libArcus/python/:../Uranium/. python3 cura_app.py 
2015-09-04 11:36:06,459 - INFO - Loaded plugin ConsoleLogger
2015-09-04 11:36:06,461 - INFO - Loaded plugin STLWriter
2015-09-04 11:36:06,462 - INFO - Loaded plugin OBJWriter
2015-09-04 11:36:06,464 - INFO - Loaded plugin OBJReader
2015-09-04 11:36:06,466 - INFO - Loaded plugin STLReader
2015-09-04 11:36:06,466 - WARNING - Plugin ConsoleLogger was already loaded
2015-09-04 11:36:06,468 - INFO - Loaded plugin WireframeView
2015-09-04 11:36:06,471 - INFO - Loaded plugin MeshView
2015-09-04 11:36:06,477 - INFO - Loaded plugin LocalFileOutputDevice
2015-09-04 11:36:06,499 - INFO - Checking for new version of cura
2015-09-04 11:36:06,499 - INFO - Loaded plugin UpdateChecker
2015-09-04 11:36:06,504 - INFO - Loaded plugin CameraTool
2015-09-04 11:36:06,516 - INFO - Loaded plugin RotateTool
2015-09-04 11:36:06,520 - INFO - Loaded plugin MirrorTool
2015-09-04 11:36:06,526 - INFO - Loaded plugin ScaleTool
2015-09-04 11:36:06,530 - INFO - Loaded plugin PerObjectSettingsTool
2015-09-04 11:36:06,533 - INFO - Loaded plugin TranslateTool
2015-09-04 11:36:06,535 - INFO - Loaded plugin SelectionTool
2015-09-04 11:36:06,539 - INFO - Loaded plugin FileLogger
2015-09-04 11:36:06,541 - INFO - Loaded plugin ChangeLogPlugin
2015-09-04 11:36:06,544 - INFO - Loaded plugin GCodeWriter
2015-09-04 11:36:06,547 - INFO - Loaded plugin LayerView
2015-09-04 11:36:06,551 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-09-04 11:36:06,560 - INFO - Loaded plugin 3MFReader
2015-09-04 11:36:06,564 - INFO - Loaded plugin SliceInfoPlugin
2015-09-04 11:36:06,571 - INFO - Loaded plugin USBPrinting
2015-09-04 11:36:06,688 - INFO - Loaded plugin CuraEngineBackend
2015-09-04 11:36:06,778 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
2015-09-04 11:36:06,779 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
2015-09-04 11:36:06,779 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
2015-09-04 11:36:06,779 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
2015-09-04 11:36:07,836 - DEBUG - Loaded theme /home/tim/Development/Cura/cura/../resources/themes/cura
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
2015-09-04 11:36:08,420 - INFO - Started engine process: /home/tim/Development/CuraEngine/build/CuraEngine
2015-09-04 11:36:08,425 - DEBUG - Backend connected on port 49675
2015-09-04 11:36:13,869 - DEBUG - Loaded a mesh with 312234 vertices
2015-09-04 11:36:25,609 - DEBUG - Killing engine process
2015-09-04 11:36:26,164 - INFO - Backend crashed or closed. Restarting...
2015-09-04 11:36:26,444 - INFO - Started engine process: /home/tim/Development/CuraEngine/build/CuraEngine
2015-09-04 11:36:26,464 - DEBUG - Backend connected on port 49675

2015-09-04 11:36:49,104 - ERROR - [Errno 4] Interrupted system call
2015-09-04 11:36:53,913 - DEBUG - Writing to Local File /home/tim/Documents/3D_models/gcode/rabbit_x2_normal_2.gcode in text mode

nallath, 2015-09-04:
Confirmed; Issue also occurs with me.

nallath, 2015-09-16:
This seems to work in current 15.10

closed 2015-09-16 by nallath