Ultimaker/Cura #325: Per object settings

BagelOrb, 2015-09-04:
I cannot get per object settings to work and I don't understand how it works.

Steps I've tried
---
- set infill to 100% in the settings, i.e. in High Quality (Customized)
- select per object settings tool
- click (+) under model
- select High Quality instead of High Quality (Customized)
- ***no difference in infill percentage in the gcode***
- select High Quality (Customized) in the other model
- ***no difference in infill percentage in the gcode***
- select Override Profile on the first (What does this do?!?!)
- click on infill density
- ***no difference in infill percentage in the gcode***
- select override profile on the other (custom)
- click on infill density (it displays 20% instead of 100%)
- ***no difference in infill percentage in the gcode***

[image]

Problems
----
1. what does override profile do? does it override the settings profile you choose for that specific model?
2. I can only select top level settings when picking a setting to override
3. what does a setting override do? I am only allowed to add a setting as override, but cannot change it. When I hover the text box a button [ x ] appears which only let's me delete the overridden setting.
4. I cannot save a profile, i.e. there is no [save as...] for profiles
5. I cannot rename a profile. The box is grayed out. 
6. I cannot manage profiles a couple of times. LOG: Segmentation fault (core dumped)
7. This happened: LOG: file:///home/tim/Development/Uranium/plugins/Tools/PerObjectSettingsTool/PerObjectSettingsPanel.qml:21:5: QML Item: Binding loop detected for property "height"
8. I cannot export profiles; after selecting a filename, nothing happens. LOG: file:///home/tim/Development/Uranium/UM/Qt/qml/UM/Preferences/ProfilesPage.qml:63: TypeError: Cannot read property 'name' of null

nallath, 2015-09-04:
Most of these will need to be answered by @awhiemstra.

awhiemstra, 2015-09-07:
Note that the UI styling for per object settings is not yet done.

"what does override profile do? does it override the settings profile you choose for that specific model?"

Override specific settings of the profile.

"I can only select top level settings when picking a setting to override"

You can select the same settings as those that are visible in the sidebar.

"what does a setting override do? I am only allowed to add a setting as override, but cannot change it. When I hover the text box a button [ x ] appears which only let's me delete the overridden setting."

The text field should be clickable but I think some details got lost due to the design changes in the sidebar, see the note above.

"I cannot save a profile, i.e. there is no [save as...] for profiles
I cannot rename a profile. The box is grayed out.
I cannot manage profiles a couple of times. LOG: Segmentation fault (core dumped)
This happened: LOG: file:///home/tim/Development/Uranium/plugins/Tools/PerObjectSettingsTool/PerObjectSettingsPanel.qml:21:5: QML Item: Binding loop detected for property "height"
I cannot export profiles; after selecting a filename, nothing happens. LOG: file:///home/tim/Development/Uranium/UM/Qt/qml/UM/Preferences/ProfilesPage.qml:63: TypeError: Cannot read property 'name' of null"

These are all related to profile management and not per-object settings. Please put them in a different issue.

BagelOrb, 2015-09-07:
Just to be clear, I didn't want my question answered - I want that such questions don't arise in users. The issue is lack of clarity and/or explanation of the term 'override' etc. 

We had discussed this fact already but I cannot recall which term we agreed upon to use instead.

As to the profile manager: https://github.com/Ultimaker/Cura/issues/342

PS: great job on the overall per-object settings work!

closed 2015-09-15 by nallath