Ultimaker/Cura #327: [15.09.80] Python crashes when switching to Simple Mode

fieldOfView, 2015-09-04:
Steps to reproduce:
* delete cura profile
* start cura, add machine profile
* switch between Simple Mode and Advanced mode
=> no problem, works as expected
* Select Normal Quality from Global Profiles
* Switch from Advanced Mode to Simple Mode
=> Cura.exe (or Python.exe when running cura_app.py) crashes ("Cura.exe has stopped working, Windows is checking for a solution")

Unfortunately neither Cura.log nor the console output (when running cura_app.py) shows anything just prior to the crash.

Win7/64bit

nallath, 2015-09-07:
Confirmed for Ubuntu 14.10. It looks like something inside of QT causes the crash.

nallath, 2015-09-07:
Interestingly enough it only happens when you added a new machine and no machine was active before.

closed 2015-09-07 by nallath

fieldOfView, 2015-09-07:
Can confirm. Source: it works.