Ultimaker/Cura #328: Toolset issue's (On Mac osx 10.10.5)

mennoberg, 2015-09-04:
- Tool button overall are too small 
- Clickability on tools without having a model, should stay passive when no model is loaded
- Icons are blurry, vector design, something goes wrong in rendering these icons in interface?
- Missing the load file icon

When model loaded:
- Tool titles are missing when tool is selected
- per object settings, what is that little square next to the tool?
- per object settings, That '+; in the circle on the model is off center, to small and doesn't respond to a hover.
- per object settings, tool on model itself is floating bellow the model, doesn't feel that it is on the model

Don't agree "override profile" function because it's very confusing, instead make profiles, easier to share and control from usability perspective. besides that:
- per object settings,  shouldn't be usable in simple mode, it's confusing
- per object settings, override profile function shouldn't with a dirty modal windows, instead select a function from the (advanced) side bar or don't use at all.

- view mode layer ain't showing anything

mennoberg, 2015-09-04:
When hover view button text hovers behind print setup bar

Ghostkeeper, 2015-09-05:
The blurry icons was found by @TMHogenhout to be caused by the SVGs having a huge viewport.

BagelOrb, 2017-07-21:
This issue is outdated

closed 2017-07-21 by BagelOrb