Ultimaker/Cura #336: Scaling disfunct after resizing a too large model

Ghostkeeper, 2015-09-05:
Version: 15.09.80
Platform: Windows 7 and OSX
Qt: 5.4.1 (on Windows 7 at least)
PyQt: 5.4.1 (on Windows 7 at least)

Steps to reproduce:

* Load a model that is too large for the printer to print.
* Click on the scaling tool and try to change some of the values.

Expected behaviour:

* The scaling tool already displays the scaling applied automatically to the model as percentages, e.g. 85%, before any of the numbers are changed.
* The scaling tool displays the current dimensions of the model.
* Changing the numbers on the scaling tool will resize the model.

Observed behaviour:

* The scaling tool says that the current scale is 100%, even though a message had popped up that it had been scaled down to e.g. 85%.
* The scaling tool displays that the model is 0 by 0 by 0 millimetres.
* Changing any of the values on the scaling tool has no effect on the model. Closing the scaling tool and re-opening it will show the dimensions reset as if they have never been changed.

nallath, 2015-09-07:
Can' t reproduce this. Could you check again with latest version?

Ghostkeeper, 2015-09-08:
Not any time soon, since building anything with PyQt on my Windows 7 laptop is hell. I could try on my Windows 8 desktop, but perhaps I'll wait for @awhiemstra to do the hard work for me in the next beta build :yum:

Ghostkeeper, 2015-09-09:
I tried to reproduce this again last night and it turns out that this was due to my own stupidity. I didn't select the model after loading.

I'm going to suggest two enhancements based on my mistake, that would improve usability: Auto-select the new model after it is loaded in Cura, and disable the textboxes/buttons of scaling, mirror, per-object settings and rotation if no model is selected. I'm going to make two new issues for these, and close this one.

closed 2015-09-09 by Ghostkeeper