Ultimaker/Cura #337: Clicking on the telemetry sending message

Ghostkeeper, 2015-09-05:
Something that improves usability a bit, and is very common and intuitive: Clicking on a message takes you to an interface that has something to do with the message. In this case:

Version: 15.09.80
Platform: Windows 7
Qt: 5.4.1
PyQt: 5.4.1

Steps to reproduce:

* Delete your settings.
* Start Cura.
* A message appears that Cura automatically sends slicing info. Click on the body of the message.

Expected behaviour:

* The preferences menu is displayed where the option is to disable sending slicing info.

Observed behaviour:

* Nothing happens.

The same could be applied to other messages, such as that the model has automatically been resized when it was too big to be printed; open the scale tool, etc.

fieldOfView, 2015-09-06:
I like the idea. It will need some good thinking though, eg: should the aforementioned "the model has automatically been resized when it was too big to be printed" message link to the scale-tool or to the preference where this auto-sizing can be turned off?

Ghostkeeper, 2015-09-06:
Scale tool seems most intuitive to me, but that'd have to be evaluated individually for each type of message.

BagelOrb, 2017-07-21:
This idea is pretty okay, but I'm closing this because of overclouding in the Cura issues.

Please reopen if you don't agree.

Ghostkeeper, 2017-07-27:
I don't really think that overcrowding the github Cura issues is a problem. There's a search function to find stuff easily, and closing issues that aren't fixed doesn't help when you use the search function. We don't need to have an overview of stuff that's left to do here since we have our Jira tracker for that.

nallath, 2017-07-27:
Automagically sending crash reports somewhere could prove to be usefull.

onitake, 2017-12-04:
Sounds like a good idea to me, though I'd generally prefer a button instead. Makes the intent clearer and the message harder to miss.

nallath, 2017-12-05:
Doesn't the 3.1 beta already have a crash log sender?

BagelOrb, 2017-12-05:
I don't see how that's related to the issue.

ianpaschal, 2018-03-09:
@Ghostkeeper Newer versions have a button that takes you directly to the telemetry preference. Is this good enough to consider this closed, or were you thinking it's good if the actual message box is one giant button?

onitake, 2018-03-17:
@ianpaschal The only thing that bothers me a bit is that the link/button only opens the settings dialog but doesn't scroll down and/or highlight the setting. If you could implement that as well, it would be perfect. But I'm fine either way.

ianpaschal, 2018-03-17:
It probably could be but I don't think it's important enough that we'll ever do it.

ianpaschal, 2018-04-04:
I'm actually closing this ticket as the discussion can be continued in @onitake's other thread: #2810.

closed 2018-04-04 by ianpaschal