Ultimaker/Cura #338: 15.09.80: Setting dialogs linked

CCS86, 2015-09-06:
Application Version: 15.09.80

Steps to Reproduce:
-Enable "Infill density" and "infill thickness"
-Set infill density
-Set infill thickness

Actual Results:
-Value entered for infill thickness also populates infill density

Expected results:
-Value only populates the highlighted field.

This is also happening in other related fields. Luckily it doesn't happen in reverse, so it is possible to get them set correctly.

CCS86, 2015-09-06:
Setting Z-hop distance overwrites printing temperature as well.

CCS86, 2015-09-06:
Clearing the "spiralize" setting also clears the "print sequence" field

nallath, 2015-09-07:
Confirmed the issue.

nallath, 2015-09-07:
Hmm, It seems to always change the first setting of a category.

closed 2015-09-07 by nallath