Ultimaker/Cura #340: 15.09.80: False Error Value

CCS86, 2015-09-06:
Cura seems to be falsely reporting a top/bottom thickness value as an error here. I have tried acount for the initial layer height, and not accounting for it:

[image]

[image]

nallath, 2015-09-07:
Confirmed.

BagelOrb, 2015-09-07:
In the json it says: "max_value_warning": "1",
There shouldn't be a max value warning.

When infill is set to 100%, the bottom thickness is set to 999999 and the top thickness to 0. This way all infill is replaced by skin. Just to give you an idea why there shouldn't be a max value warning.

closed 2015-09-17 by nallath