Ultimaker/Cura #341: 15.09.80: Layout obscuring unit values

CCS86, 2015-09-06:
[image]

nallath, 2015-09-07:
A known issue. Thanks for reporting it.

fieldOfView, 2015-09-07:
Instead of using a "R(eset)" button, a UI pattern for showing changes in settings I know from other applications (Firefox, Cinema4d, probably others) is to render the setting in bold. Double-click (and/or a context menu) to reset to the default setting.

nallath, 2015-09-15:
Fixed for 15.09.82

closed 2015-09-15 by nallath

fieldOfView, 2015-09-15:
@Nallath: I would not necessarily agree the issue is fixed (though the button no longer overlays the units); now the [R] button is obscured by the scrollbar...

nallath, 2015-09-15:
Technically that would make it a new issue. But you're right. We should fix that aswell (hadn't noticed that yet)

CCS86, 2015-09-15:
Hey Jaime, will we see .82 on the beta site any time soon?

awhiemstra, 2015-09-15:
@CCS86 Probably thursday