Ultimaker/Cura #342: 15.06.2: Manage Profiles Bugs

CCS86, 2015-09-06:
Steps taken:

-Select a global profile
-Manage settings to enable additional settings
-Customize settings
-Manage profiles, select customized profile, options to remove or rename are disabled. Choose export and pick a file name: no error, no file written.
-Load a model, let it slice
-Choose manage profiles: Cura crashes

nallath, 2015-09-07:
Could you post a log of where this happens?

nallath, 2015-09-07:
Confirmed the issue. Did fix another issue that I thought might be the cause of this (but isn't).

BagelOrb, 2015-09-07:
From another isse, I am expanding the list of problems to the following:
1- I cannot save a profile, i.e. there is no [save as...] for profiles
2- Saving profiles doesn't work "no error, no file written."
3- I cannot rename a profile. The box is grayed out.
4- I cannot manage profiles a couple of times. LOG: Segmentation fault (core dumped)
5- This happened: LOG: file:///home/tim/Development/Uranium/plugins/Tools/PerObjectSettingsTool/PerObjectSettingsPanel.qml:21:5: QML Item: Binding loop detected for property "height"
I cannot export profiles; after selecting a filename, nothing happens. LOG: file:///home/tim/Development/Uranium/UM/Qt/qml/UM/Preferences/ProfilesPage.qml:63: TypeError: Cannot read property 'name' of null

nallath, 2015-09-07:
You're looking at master branch, not the 15.10 branch. I've fixed a number of those issues there.

BagelOrb, 2015-09-07:
I first reported these issues at the first time of release, this was the last day in which there was no 15.10 branch yet ;)

nallath, 2015-09-07:
Some other change crash that i fixed also seems to fix this.

closed 2015-09-15 by nallath