Ultimaker/Cura #343: 15.06.2: Not Obeying Shell thickness

CCS86, 2015-09-06:
-Load model
-Select normal profile
-Change shell thickness to 2mm

Cura slices and displays a 0.8mm shell thickness. Checked gcode, and it is the same.

[image]

CCS86, 2015-09-06:
Not sure if this is front end or engine related

nallath, 2015-09-07:
I can't reproduce this. I get a correct model. Could you check if the right settings are in the g-code? At the bottom you should find human readable settings.

BagelOrb, 2015-09-07:
Most probably frontend. 

In the settings json file the wall line count is computed as such:
max(1, (int(parent_value / (machine_nozzle_size - 0.0001) + 1) **if** (parent_value / max(1, int(parent_value / (machine_nozzle_size - 0.0001))) > machine_nozzle_size) * 1.5 **else** int(parent_value / (machine_nozzle_size - 0.0001))))

Certainly there is a problem with the following pattern: **if** (...  > ...) * 1.5 **else** 

Moreover, I'm not sure whether the formula is correct at all. A shell thickness of 2.4 should lead to a wall thickness of 2.4, in which case `max(1, (int(parent_value / (machine_nozzle_size - 0.0001) + 1)`evaluates to 7 instead of 6!

CCS86, 2015-09-11:
@nallath , Sorry it took so long for me to confirm. 

I tested again today with 15.09.81 and found the same issue. The posted gcode does list the correct value:

https://dl.dropboxusercontent.com/u/60958586/15.09.81.gcode

[image]

nallath, 2015-09-11:
Then I think it's an engine issue, as the interface sends the right value.

CCS86, 2015-09-11:
That makes sense. Do we need to post the issue over there, or is the CuraEngine label enough?

nallath, 2015-09-12:
The label should be enough, but @BagelOrb  is on a holiday now, so this will take a bit of time until it's picked up.

CCS86, 2015-09-12:
Did some more testing and found front-end issues too.

It is possible to enable both "shell thickness" and "wall line thickness/count". They do not stay linked, so it is possible to set them with conflicting values. I think that only one of these options should appear at a time, disabling the parent option.

[image]

It is also possible to set the shell thickness or wall thickness to values not divisible by nozzle width, without a warning:

[image]

CCS86, 2015-09-12:
With only wall line count enabled, it will slice at the default value of 2, but slicing will not happen with any other value that I have tried. 

Not sure if that is engine or front end.

CCS86, 2015-09-21:
I am now able to get thicker shells using the wall line count setting. Shell thickness still doesn't work, and might actually cause a crash when set to conflicting values.

BagelOrb, 2015-09-29:
@nallath maybe you overlooked my comment above which explains what exactly is wrong in the front-end.

Moreover, in the gcode is says 
```-s wall_line_count="2"```

This is no backend issue, I only see the frontend issues I've listed above.

BagelOrb, 2015-09-29:
The new function for computing the wall_line_count is now:
    max(1, round((wall_thickness - wall_line_width_0) / wall_line_width_x) + 1)

The initial issue seems solved.

However, another issue seems to cause the inheritance of wall_thickness from shell_thickness to fail.

closed 2015-09-29 by BagelOrb

BagelOrb, 2015-09-29:
See https://github.com/Ultimaker/Cura/issues/442