Ultimaker/Cura #344: Crash! Version: 15.06.03 uncaught exception happened

TomDeng9436, 2015-09-06:
Version: 15.06.03
Platform: Windows-8-6.2.9200
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 124, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 937, in __getitem__
KeyError: 'general'

nallath, 2015-09-07:
Remove the preferences file.

closed 2015-09-07 by nallath

TomDeng9436, 2015-09-07:
No it`s still not working after I remove the files.

code 

Version: 15.06.03 

Platform: Windows-8-6.2.9200

Qt: 5.4.1

PyQt: 5.4.1

Exception:

Traceback (most recent call last):

File "cura_app.py", line 9, in <module>

File "c:\dev\uranium\UM\Application.py", line 329, in getInstance

File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__

File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__

File "c:\dev\uranium\UM\Application.py", line 67, in __init__

File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile

File "c:\dev\uranium\UM\Preferences.py", line 124, in _loadFile

File "c:\dev\python34\lib\configparser.py", line 937, in __getitem__

KeyError: 'general'

nallath, 2015-09-07:
This sounds exactly like an error that we had before (where removing the file fixed it). Could you add a line to the file containing: [general]