Ultimaker/Cura #346: Cura.cfg error

ripper121, 2015-09-07:
I had opened multiple windows from cura, after closing this the cfg file has errors.

Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 122, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1088, in _read
configparser.ParsingError: Source contains parsing errors: 'C:\\Users\\STS/AppData/Local/cura\\cura.cfg'
	[line 18]: 'iew]\n'

Cura.cfg:

[general]
language = de
window_width = 1920
window_height = 1028
version = 2

[view]
show_overhang = True

[cura]
categories_expanded = ;cooling;infill;material;platform_adhesion;resolution;speed;support
active_mode = 1
recent_files = C:/Users/STS/Documents/Sketchup/Termostat/Cov-in T4 10-R closed.STL;C:/Users/STS/Documents/Sketchup/Termostat/Cov-in T4 10-L.STL;C:/Users/STS/AppData/Local/Temp/Cov-in T4 10-L.STL;C:/Users/STS/Documents/Sketchup/Rasp_Body_Lowerpart.stl;C:/Users/STS/Google Drive/Arbeit/WallCase/WallCase_6_PCB.stl;C:/Users/STS/Google Drive/Arbeit/WallCase/WallCase_6_Therm_Top_0.stl;C:/Users/STS/Google Drive/Arbeit/WallCase/WallCase_6_Top.stl;C:/Users/STS/Google Drive/Arbeit/WallCase/WallCase_6_SF_Sample1.stl;C:/Users/STS/Google Drive/Arbeit/WallCase/WallCase_6_SF_Sample0.stl;C:/Users/STS/Google Drive/Arbeit/WallCase/WallCase_6_Therm_Top.stl;
active_machine = Ultimaker 2

[backend]

iew]
show_overhang = True

nallath, 2015-09-07:
Thanks for reporting it, this really help us. We've had a ton of problems with the config before, but your data really helps narrowing it down (as it has very specific behaviour :))

nallath, 2015-09-23:
This should be fixed in the next version.

closed 2015-09-23 by nallath