Ultimaker/Cura #348: 15.09.8 Release notes & Changelog

BagelOrb, 2015-09-07:
New features (since 15.06) not mentioned:
- Fuzzy skin mode (Something went wrong here, the feature has been commited on master of both repos, but at that time the frontend hadn't created the 15.10 branch)
- Z-seam alignment

In the changelog it says:
‘Follow Mesh Surface’ has been removed from the engine, the same result can be 
achieved using no infill or top/bottom layers.

Both these statements are false. 

However, mesh surface mode is currently not fully working for meshes with disconnected faces. Perhaps it should be removed, since the bugfixing is not as easy as it seems.

The changelog also mentions July instead of September/Oktober: "Available to download for free from the xxth of July, new Cura is set to take the 3D printing experience to a whole new level. "

SEADEMON, 2015-09-07:
There autoreduction model with huge dimensions. Exhibited scale 1 but the size is wrong. And you could add zoom in on a like the old version is not % .

nallath, 2015-09-08:
@SEADEMON Please put seperate things in a seperate issues. It helps us keep track of it. 

I'm going to leave this to @awhiemstra as he wrote the change log together with marketing.

BagelOrb, 2015-09-08:
Fuzzy skin mode is now in the 15.10 CuraEngine as well

nallath, 2015-09-10:
The changelog is outdated. I've added the two features to the changelog plugin.

closed 2015-09-10 by nallath