Ultimaker/Cura #352: Autorescale not rescaling

BagelOrb, 2015-09-08:
The object is larger than the print volume, bu the scaling is 100% and the reported scaling is 0%!

[image]

nallath, 2015-09-08:
It works for my test model. Could you send me yours?

awhiemstra, 2015-09-08:
It looks like a 20x20x20 or similar cube. This probably confuses the scaling code as there is no clear "maximum" axis.

BagelOrb, 2015-09-08:
It is.

Here! :P 

solid OpenSCAD_Model
  facet normal -0 0 1
    outer loop
      vertex 0 100000 100000
      vertex 100000 0 100000
      vertex 100000 100000 100000
    endloop
  endfacet
  facet normal 0 0 1
    outer loop
      vertex 100000 0 100000
      vertex 0 100000 100000
      vertex 0 0 100000
    endloop
  endfacet
  facet normal 0 0 -1
    outer loop
      vertex 0 0 0
      vertex 100000 100000 0
      vertex 100000 0 0
    endloop
  endfacet
  facet normal -0 0 -1
    outer loop
      vertex 100000 100000 0
      vertex 0 0 0
      vertex 0 100000 0
    endloop
  endfacet
  facet normal 0 -1 0
    outer loop
      vertex 0 0 0
      vertex 100000 0 100000
      vertex 0 0 100000
    endloop
  endfacet
  facet normal 0 -1 -0
    outer loop
      vertex 100000 0 100000
      vertex 0 0 0
      vertex 100000 0 0
    endloop
  endfacet
  facet normal 1 -0 0
    outer loop
      vertex 100000 0 100000
      vertex 100000 100000 0
      vertex 100000 100000 100000
    endloop
  endfacet
  facet normal 1 0 0
    outer loop
      vertex 100000 100000 0
      vertex 100000 0 100000
      vertex 100000 0 0
    endloop
  endfacet
  facet normal 0 1 -0
    outer loop
      vertex 100000 100000 0
      vertex 0 100000 100000
      vertex 100000 100000 100000
    endloop
  endfacet
  facet normal 0 1 0
    outer loop
      vertex 0 100000 100000
      vertex 100000 100000 0
      vertex 0 100000 0
    endloop
  endfacet
  facet normal -1 0 0
    outer loop
      vertex 0 0 0
      vertex 0 100000 100000
      vertex 0 100000 0
    endloop
  endfacet
  facet normal -1 -0 0
    outer loop
      vertex 0 100000 100000
      vertex 0 0 0
      vertex 0 0 100000
    endloop
  endfacet
endsolid OpenSCAD_Model

nallath, 2015-09-08:
The reported scaling is rounded (the scale does happen correctly!). It does not take the disallowed areas into account.

BagelOrb, 2015-09-08:
The reported scaling is the actual scaling, you are right.

However, the scaling did not happen correctly.

Actual: model is higher than build volume.
Expected: model just fits within build volume.

nallath, 2015-09-09:
Weirdly enough, it works for me. Do you get this issue consistantly? Or with all objects?

BagelOrb, 2015-09-09:
I get it consistently with that model. Can you provide me with other huge models? Then I could check.

nallath, 2015-09-09:
I just made a huge box with Blender, which worked. I'm sending you an email with the files.

BagelOrb, 2015-09-09:
[image]
[image]

It seems they are scaled to fit in the X and Y dimension, but that leaves Z too high.

nallath, 2015-09-09:
Ah, nice catch. You're right. I'll look into it.

closed 2015-09-09 by nallath