Ultimaker/Cura #354: Bug: merge keeps objects on platform

BagelOrb, 2015-09-08:
Merge keeps each object on platform instead of combination of all.

Actual:
[image]

Expected:
[image]

nallath, 2015-09-09:
Do you have a download link of that object?

BagelOrb, 2015-09-09:
https://www.youmagine.com/designs/guy

BagelOrb, 2015-09-09:
The combined object should still fall on the platform.
[image]

nallath, 2015-09-09:
According to the bounding box it is ;) It seems like the bounding box is calculated wrong. I'm not able to reproduce this. Could you send me the file where this happens?

BagelOrb, 2015-09-09:
Its just two cubes translated up in openscad...

BagelOrb, 2015-09-09:
* translate([0,0,10]) cube([10,10,10]);
translate([0,0,20]) cube([10,10,10]);

closed 2015-09-10 by nallath