Ultimaker/Cura #356: Auto-select model after loading

Ghostkeeper, 2015-09-09:
When a model is loaded, it seems logical to me to select it immediately, such that the scaling, rotation, etc. tools can immediately be used.

What are your thoughts on this?

nallath, 2015-09-09:
But what should be the behaviour when another model is already selected? Add it to the selection or reset the selection so the one object is selected?

Ghostkeeper, 2015-09-09:
No change in selection could also be considered in that case. I'd go with either no change in selection or to reset the selection.

nallath, 2015-09-09:
@fieldOfView @TMHogenhout @mennoberg @awhiemstra  Any takes on this?

fieldOfView, 2015-09-09:
Something to consider here is what legacy-Cura does. Not because legacy-Cura is holy, but because it is what a sizeable portion of your users will expect. I just checked, and it does (re)set the selection to just the newly loaded object.

Next I tried Cinema4D, which also sets the selection to a newly created or merged in loaded object, as I would expect most 3d packages to behave (but I don't have many to confirm this with). It does make sense to me.

My vote is for "(re)set the selection to just the newly loaded models". I added the plural in case a user selects multiple files in the open dialog, or drags in multiple files at the same time. Don't know if that is supported.

awhiemstra, 2015-09-09:
Personally, I'd be more inclined to not change the selection if there already is a selection, this with the underlying assumption that suddenly changing things can mess up what people are doing.

@fieldOfView Actually, dragging is the only way to load multiple objects at the same time at the moment, because I had some problems with file dialogs and had to disable multi selection for them.

fieldOfView, 2015-09-09:
>  this with the underlying assumption that suddenly changing things can mess up what people are doing.

What they are doing is adding one or more new objects. So I would argue that the change is not "sudden"/unexpected. Setting selection to the new objects has the benefit of being able to press delete/backspace to remove the object(s) again.

nallath, 2015-09-09:
I think it makes sense to change the selection. Loading a new model is always an user initiated action.

mennoberg, 2015-09-09:
I Like the fact that tools are always visible, whether a model is loaded or not. But I can see it might be confusing when using a tool nothing happens to nothing. A more layered view on those tools, a 50% transparency and unable to hover might solve that confusion. When a model is selected the tool response as you expect.

Another option is when you select a tool, Cura ask you to select an model. 

Both option are clear and friendly in my opinion, though the first might be more intuitive where in the second option you're more guiding a user, which is a thing you try to avoid a bit I guess.

awhiemstra, 2015-09-09:
@fieldOfView I'm fine with either and do not really have a strong opinion, it just felt slightly better to me. You're right that the action is opening new files, though I would recommend using Ctrl + Z to undo the loading. ;)

BagelOrb, 2017-07-21:
This still seems like an easy feat which is pretty nice.

ianpaschal, 2018-05-22:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Won’t Do` and closing the issue.

closed 2018-05-22 by ianpaschal

ianpaschal, 2018-05-23:
Big thank you to Mr. FieldOfView for taking this off our plate and implementing it himself.

PR #3848 (2018-05-23): Select models after loading
This PR selects models after loading them into Cura. This makes Cura behave like other applications, including legacy Cura.

Fixes #356
commit f6726148 (2018-05-23): Select models after loading

Fixes #356