Ultimaker/Cura #357: Disable tools when no model is selected

Ghostkeeper, 2015-09-09:
If no model is selected, the tools are not functional, but they can still be interacted with. I think it would be good to disable these tools (where applicable) if no model is selected.

Since these tools are implemented as plug-ins, I suggest that this would be realised in the plug-in itself, rather than making it inherent to tools in general. I can imagine there are use cases where tools aren't disabled when no model is selected, such as an align-to-grid button or a remove-all button.

nallath, 2015-09-09:
Good point. This would require that tools have an 'enabled' state of some sorts that allows them to disable themselves. This does mean it's going to be in the 16.01 release due to feature freeze for 15.10

awhiemstra, 2015-09-09:
The tricky thing is that you basically have three states: No selection, one object selected and several objects selected. No selection/something selected we can probably consider a bugfix since it is mostly visualisation of a state that is already implied. When we add multiple objects selected we can then implement buttons for grouping/merging to make those things more visible.

Ghostkeeper, 2015-09-10:
Menno mentioned this issue in #356:

> I Like the fact that tools are always visible, whether a model is loaded or not. But I can see it might be confusing when using a tool nothing happens to nothing. A more layered view on those tools, a 50% transparency and unable to hover might solve that confusion. When a model is selected the tool response as you expect.

> Another option is when you select a tool, Cura ask you to select an model.

> Both option are clear and friendly in my opinion, though the first might be more intuitive where in the second option you're more guiding a user, which is a thing you try to avoid a bit I guess.

I don't think making them invisible is a good idea either, but perhaps making the background grey and disabling interaction is enough.

TMHogenhout, 2015-10-06:
I added an inactive state to the toolbuttons and the viewmode button and I greyed them out because I found it clearer and it's visually a bit calmer. 

https://github.com/Ultimaker/Cura/commit/c3036101f46e092c615fbcab43dd0dbc572e0d3f

But this is only tied to whether there is platform activity (is there something on the buildplate?). I agree that activating the buttons when 1 or more meshes are selected, would be better.

awhiemstra, 2015-10-07:
And I later fixer it to use the selection and removed the disabling of the view button since that was causing more issues than it solved.

closed 2015-10-07 by awhiemstra

mennoberg, 2015-10-07:
Request for demo and team-review