Ultimaker/Cura #358: Pan to model occurring too often

BagelOrb, 2015-09-09:
When I want to drag around objects everytime I start dragging another object, the view pans to center on the object, which causes the object to move, because I am dragging, while I am not moving my mouse.

It would be nice if the view panning would only occur if I click the object (and let go within t seconds and/or within x distance moved with the mouse), not when I start dragging the object.

Ghostkeeper, 2015-09-09:
Or maybe if you release the mouse button?

fieldOfView, 2015-09-09:
I think a good solution would be to only center the camera on an object if the object is clicked (ie: press + release in a short time), instead of when the selection changes. This way you can drag objects around without chaging the "focus" all the time.

BagelOrb, 2015-09-09:
The suggestion by @Ghostkeeper is a nice quick fix for the current beta release though ;)

nallath, 2015-09-09:
You can set the 'Center camera when object is selected' preference to false as a quick work around. I'm not quite sure if we can see the difference between a click and a longpress easily.

BagelOrb, 2015-09-09:
As I said, as a quick fix you might implement Ghostkeepers idea. You should be able to see the difference between an IS_CLICKED event and an IS_RELEASED event...

nallath, 2015-09-10:
Yeah, but currently we look at a selection changed event. Not at a mouse event. For instance; we also think about implementing a selection change when an object is loaded. In this case i would want the pan to happen.

BagelOrb, 2015-09-10:
Then maybe dragging an object shouldn't result in a selection change...

fieldOfView, 2016-11-09:
Issue can be closed

closed 2016-11-09 by nallath