Ultimaker/Cura #360: Bug: Scale + ungroup

BagelOrb, 2015-09-10:
Steps:
- **merge** multiple objects
- scale merged group
- ungroup

Expected: ungrouped items are scaled
Actual: returned to original size

nallath, 2015-09-10:
This is by design. So i'm gonna use the help lines of @TMHogenhout  and @fieldOfView to help decide on this.

BagelOrb, 2015-09-10:
I have no idea whatsoever why you should want this behaviour. Can anyone elaborate?

fieldOfView, 2015-09-10:
I agree with @bagelorb. I would not expect ungrouping objects changes their scale, just like I don't expect the object to magically conform to the group scale when I add the object to the group. (Yeah, I know, from the application viewpoint it is the other way around; if you want to preserve the "inherited" scale, you have to do some magic when you ungroup). 

In all 3d packages I have worked with, ungrouping an object applies the group transformation to the object before reparenting.

BagelOrb, 2015-09-10:
I think that other-way-around thinking is what confused people into thinking the current behaviour was the intended behaviour.

@fieldOfView Your clarification is enlightening!

nallath, 2015-09-10:
I was leaning towards bagel orbs solution, but i'd rather discuss these things before implementing them.

BagelOrb, 2015-09-10:
What I am @fieldOfView are saying is the same thing.

When removing an item from a group, the group tranformations should be applied to the object, so that the object is effectively not transformed.

When adding an item to a group, the group transformations should be unapplied to the object, so that the object is effectively not transformed.

Through both these changes it holds that:
Group transformation + object transformation on grouped object = object transformation on non-grouped object.

closed 2015-09-10 by nallath

BagelOrb, 2015-09-10:
I see you've not implemented what I said should happen when adding something to a group.

This is because there is no such thing. Cura only allows you to make a new group, or explode. When you 'add' to a group you are really making a new group with one of its children being a group.

tl;dr: nvm