Ultimaker/Cura #367: [Reset to Default] not reset

BagelOrb, 2015-09-10:
Steps:
- change settings
- Select global profile High Quality (not the Customized one)
- watch all settings revert to default
- change any settings

Actual:
all previously changed settings get the [ R ] reset to default option again

Expected:
only the currently changed settings gets the [ R ]

nallath, 2015-09-10:
I think this is more an user interaction choice then an actual bug. What it does is switch back to the previously edited custom profile. What we also could do is see that there already is an changed profile and add a new one (eg; High Quality (customized) 1)

BagelOrb, 2015-09-10:
No it doesn't, the values stay the same as in the non-customized profile.

I wanted a reset to defaults option, just as in the old Cura. I found it by loading the profile High Quality, and when I edit a setting all settings stay the same, the only difference is that the [ R ]'s pop up all over all the settings.

nallath, 2015-09-10:
For me they changed.

BagelOrb, 2015-09-10:
Oh yes they changed now.

I would expect that if you change the High Quality profile, the old High Quality (Customized) will get thrown away and overwritten by the normal High Quality profile except the changed setting.

When you want to change back to your current settings, just change back to High Quality (Customized) - not by editing the High Quality profile.

An are-you-sure popup might be needed when overwriting the old High Quality (Customized)...

closed 2015-09-22 by TMHogenhout