Ultimaker/Cura #370: Bug: filename

BagelOrb, 2015-09-10:
[image]

How is this supposed to work?
I would expect the filename should correspond to the filename specified.

Should the filename field get autogenerated from the model filename?

Ghostkeeper, 2015-09-10:
You mean that the gcode file should have the same filename as the "Printjob name" specified above?

BagelOrb, 2015-09-10:
What else is Printjob name?

What's a printjob?

nallath, 2015-09-10:
printjob name doesnt work yet if i recall correctly.

awhiemstra, 2015-09-10:
It does not. We are currently still even debating the label for that item. :)

awhiemstra, 2015-09-25:
This has been implemented for the next build.

closed 2015-09-25 by awhiemstra