Ultimaker/Cura #371: minor bug: Z-fighting in layer view

BagelOrb, 2015-09-10:
The top skin is z-fighting with the 'ghost' of the object.

BTW: is it intended behaviour that the ghost is there? It seems a bit hacky, but I kinda like it! You can change the position of an object while looking at the layer view!
Can't really specify why it looks like a hack to me...

[image]

awhiemstra, 2015-09-10:
Yes, the ghost is intended. This way you do not end up with major editing workflow differences between one view and another. I guess one reason it feels a bit hacky is that you expect to change the position of the layers and not a selection ghost. However, actually changing the layers is currently technically quite tricky.

BagelOrb, 2015-09-10:
When a reslice has been triggered, the layer view should get cleared.

This to prevent the confusion some people have with changing the ghost and seeing the gcode stay at the same place.
https://ultimaker.com/en/community/view/17296-cura-1510-open-beta?page=2#reply-117818

fieldOfView, 2016-11-09:
This issue can be closed. Ghosts are now rendered in a different pass than layers, so z-fighting does no longer occur

closed 2016-11-09 by nallath