Ultimaker/Cura #372: Customizing profiles

BagelOrb, 2015-09-10:
Steps:
- load High Quality
- change some settings
- rename the current High Quality (Customized) to My_uber_settings

- load High Quality
- change some settings
- note that the profile is changed to High Quality (Customized)

- load My_uber_settings
- change some settings
- note that the profile is *not* changed to My_uber_settings (Customized)

Why the difference?
Expected: no difference; actual: difference.

I would like it if My_uber_settings (Customized) was auto-created as well!

awhiemstra, 2015-09-10:
The difference is that "My_uber_settings" is located in a writeable location and is considered an already customised profile, whereas normal/high are shipped with Cura and in a non-writeable location. The problem with doing what you say is that you can easily end up with a "High Quality (Customised)(Customised)(Customised)(Customised)" profile.

BagelOrb, 2015-09-10:
Maybe just check if the name already ends in "(Customized)"? Or to avoid the translation difficulties, just have "High Quality" and "High Quality* "

The other possibility, which is more standard, is to always switch to "Custom" whenever you change something.

However, I really like the fact that you can have your own current version of every profile you saved, even though it can mean a duplication of the whole profile list.

Whatever you choose on this subject, it should never be the case that changing a setting in a saved profile automatically changes the profile without asking. You should be able to [ save ] the profile when you want, not when you accidentally made some changes.

Ghostkeeper, 2015-09-11:
I like the * variant, by the way. It coincides with other applications I'm familiar with.

BagelOrb, 2015-09-29:
Also: instead of High Quality and High Quality* being in the same list, as soon as you change settings a star should be added and the option to reset to default should appear somewhere. 

Then the problem doesn't occur where you change a setting, go back to default, change another setting and suddenly have the first setting changed again as well, instead of it having default value.

nickthetait, 2015-09-29:
Here is a situation to consider:
Default high quality profile is selected
Customizations are made, save as a new profile "test"
More modifications are done, save as a new profile "test2", after much printing this profile is validated/approved
User deletes "test" profile

It is useful to know which custom profiles came from which default profile. Whereas the intermediate customized profiles may get deleted at some point.

A completely different system for designating user customized profiles would be to setup a categorization system in the UI (rather than just adding a * into the name).  This is what we did in v17 of LulzBot edition of Cura. Note: we are still using the old wxpython version of Cura for now. There are no user customized profiles yet, but the categories are the important part.  [Lil screenshot](https://cloud.githubusercontent.com/assets/11818614/10174263/234e34f2-66a6-11e5-947d-c46282baaa87.png) Main reason for swapping radio buttons to dropdown menus is because we have [plenty of different materials](https://cloud.githubusercontent.com/assets/11818614/10174292/49eceab8-66a6-11e5-9fe4-28d3369b5cc3.png) available now.

BagelOrb, 2017-07-21:
Most of this is implemented by now

closed 2017-07-21 by BagelOrb