Ultimaker/Cura #374: Bug: positioning and merging

BagelOrb, 2015-09-10:
Steps:
- load model 3 times
- apply some scalings
- merge two
- apply some scaling
- merge the third with it

Problems:
- origins of merged objects don't coincide

[image]

For clarity: a view from below showing there's a small dodecahedron in the big one:
[image]

Merging another time lead to:
[image]

perhaps the push-free causes the origins not to coincide anymore

nallath, 2015-09-11:
This is a weird way of doing it. Objects that are merged should be objects that are -supposed- to be merged in the first place. If you scale one of those objects, but not the other, this relation is already broken.

nallath, 2015-09-16:
We need to come up with good behaviour for this; What would be expected in this case? Would the object scale on the largest object or simply to a unity scale?

Ghostkeeper, 2017-12-04:
Our project manager just removed this one from our planning because it's older than 12 weeks.

fieldOfView, 2017-12-04:
@BagelOrb, wouldn't grouping instead of merging work here?

BagelOrb, 2017-12-04:
No, the problem is that merging groups isn't well behaved.

I would expect that performing a merge on a merged group and another model would result in the same positioning as merging all objects as once.

[image]

Still it's not a big problem because you can just ungroup all and do the merge on all objects at the same time.

[pie_slices.7z.txt](https://github.com/Ultimaker/Cura/files/1527143/pie_slices.7z.txt)

Vandrasc, 2018-04-09:
Is this still an issue? Can we close this one?

BagelOrb, 2018-04-09:
Yep it's still an issue:
[image]

It's not a big issue, though. The workaround is to unmerge the models and doing a merge on all at once.

closed 2018-11-22 by Appesteijn