Ultimaker/Cura #375: lil bug: print times don't fit

BagelOrb, 2015-09-10:
Not a big issue, but long print times can overlap with the materials info:
[image]

BagelOrb, 2015-09-10:
Moreover, I don't really like the format. I would rather see

1d 22h 11m

or

82h 11m

problems now:
- zero in front
- *d*ays have different format style than *h*ours and *m*inutes
- string can be too long for its space

BagelOrb, 2015-09-10:
another problem: the unit for time and length are both 'm' 

I think minutes should be abbreviated as 'min'.

Ghostkeeper, 2015-09-11:
Please do retain the number of days. It's important to the common use case of people that make models that take several days to print.

That said, I agree that the "22:11 h:m" part is a bit strange.

nallath, 2015-09-15:
Fixed for 15.09.82

closed 2015-09-15 by nallath