Ultimaker/Cura #376: Bug: [ Delete ] deletes wrong objects

BagelOrb, 2015-09-10:
Pressing [delete] deletes the wrong object.

First I had it deleting the other object (group), while something else was selected.
Then I got it to delete the right thing again.
Now it's not deleting anything anymore.

Please try to replicate any such bug by playing around.

Close this issue if you cannot replicate it after playing around for  quite a while.

nallath, 2015-09-14:
I've not been able to reproduce this. I did make quite a bit of changes in how grouping is handled, so i might have accidently fixed this.

BagelOrb, 2015-10-22:
This is still a problem.

I now did these steps twice and the problem reoccured.

Steps:
-Load model
-Load model 2
-Merge models
-Load model
-left click on last loaded model
- press [delete]
- watch wrong model get deleted

BagelOrb, 2015-10-22:
just before pressing delete:
[image]

BagelOrb, 2017-07-21:
Cannot reproduce

closed 2017-07-21 by BagelOrb