Ultimaker/Cura #379: Delete all objects warning

hezhao, 2015-09-10:
There is no warning for deleting all objects after right clicking on the view window, it's quit dangerous as someone can accidentally delete all objects after carefully placing them on the platform.

[image]

nallath, 2015-09-11:
The screenshot you posted is of the old Cura. The new cura doesn't have this option either.

awhiemstra, 2015-09-25:
Well "clear platform" removes all objects from the platform which basically does the same. However, since we have undo support in the new Cura this should be undoable and not be an issue.

closed 2015-09-25 by awhiemstra