Ultimaker/Cura #381: 15.09.81: Coast glitch

CCS86, 2015-09-11:
Hi guys, just installed .81 today. So far much better!

I enabled "coasting" and left the default values. The posted code shows an occasional anomaly:

[image]

BagelOrb, 2015-09-28:
This seems like it's caused by the z alignment bug, which is the result of the fact that each layer starts on a vertex and slight differences cause the vertices of different layers not to be aligned.

BagelOrb, 2015-09-30:
I'm more interested in why the skirt seems to penetrate the model. Can you maybe send me the model so that I can test on it to make it print better?

CCS86, 2015-10-05:
Sorry for the delay. The skirt actually doesn't penetrate the model. There is just a very small contact patch and an immediate overhang. I'll post the model up.

BagelOrb, 2016-03-07:
This doesn't seem like a bug per se, so I'm closing this.

closed 2016-03-07 by BagelOrb